ros-controls/ros2_control #260: Publishing of the joint_state does not follow the desired rate

buschbapti, 2020-12-08:
I have tried following the ros2_control_demos and made some testing to push the frequency of the update rate. Turned out that `hz` the `/joint_states` topic produces inconsistent behaviors with the receiving frequency sometime 10 time slower than expected. I know you are using `MultiThreadedExecutor` and to my experience I had this issue on a publisher on my own. I initially thought it was because I was using Docker and there was some configuration to do but turns out I can reproduce the same on a desktop installation.

I have opened an issue [#1487](https://github.com/ros2/rclcpp/issues/1487#) on [rclcpp repository](https://github.com/ros2/rclcpp/) because I do think it comes from the `MultiThreadedExecutor` that is not reliable. Have you ever had the same issue? Did you solve it by using some specific things like real time kernel?

buschbapti, 2020-12-10:
Ok good progress has been made on rclcpp side. It has been confirmed that this problem comes from the `MultiThreadedExecutor`  and a path has been found. I have tested it and the solution works for the demo as well and solves this issue. Do you want me to close this issue or should I keep it open until the patch has been pushed to rclcpp?

destogl, 2020-12-10:
Hi @buschbapti,
Thanks for the detailed analysis! I would keep it open for now, sine probably other people will have it also.

So it seems this will be solved in galactic?

destogl, 2020-12-20:
This is related to PR ros2/rclcpp#1168

@bmagyar: can you rename this to "MutliThreadedExecutor update rate is unstable" (or something similar?

bmagyar, 2021-02-16:
This is a duplicate of #277, closing in favour

closed 2021-02-16 by bmagyar