ros-controls/ros2_control #262: Add resource check in switch controller callback

v-lopez, 2020-12-11:
In ROS1 control the resource compatibility was checked in the switch controller callback: https://github.com/ros-controls/ros_control/blob/noetic-devel/controller_manager/src/controller_manager.cpp#L581

Currently this check does not exist, and is only performed in start_controller (https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L603) which prints an error and fails to start a controller, but the rest of the controllers are started as if nothing happened. 

Besides the log error there's no failure notification to the user.

destogl, 2020-12-12:
It seams we need L582-L594 and L597-L605 in CM.cpp to do this.
Is "switch_controller" service only way for user to start/stop controllers? if so we could move this in "switch" function. This would be probably the easiest way to provide sensible error-feedback (controller name + interface name) to users.

bmagyar, 2021-03-24:
will bundle with https://github.com/ros-controls/roadmap/pull/41 and see if anything is missing to be captured

closed 2025-07-14 by github-actions[bot]