ros-controls/ros2_control #269: Member function naming in system_interface.cpp

AndyZe, 2020-12-14:
These member functions defy the Google style guidelines and the ROS2 C++ style guidelines. Was that intentional or should we change them?

[hardware_interface/system_interface.hpp](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/system_interface.hpp)

For example, `export_state_interfaces()` should be `exportStateInterfaces()`

Google style guide:  https://google.github.io/styleguide/cppguide.html#General_Naming_Rules
ROS1 style guide: http://wiki.ros.org/CppStyleGuide#Function_.2F_Methods

I'm happy to make a cleanup PR if needed

AndyZe, 2020-12-14:
It looks like this is pretty widespread throughout the codebase

gavanderhoorn, 2020-12-14:
ROS 2 C++ code style allows `snake_case`.

From [Code style and language versions - C++ - Style: Function and Method Naming](https://index.ros.org/doc/ros2/Contributing/Code-Style-Language-Versions/#function-and-method-naming):

> [image]

AndyZe, 2020-12-14:
Cool. Maintainers, feel free to close the issue

closed 2020-12-14 by AndyZe