ros-controls/ros2_control #274: Build Error: Unable to extract the library file path from "" generated when building

DrFKNoble, 2020-12-17:
Hi there,

I'm trying to test out ros-control's ros2_control_demo's; however, when building the ros2_control repo, I get the following error:

```console
Starting >>> hardware_interface
--- stderr: hardware_interface                         
CMake Error at /opt/ros/foxy/share/tinyxml2_vendor/cmake/tinyxml2_vendor-extras.cmake:23 (message):
  Unable to extract the library file path from
Call Stack (most recent call first):
  /opt/ros/foxy/share/tinyxml2_vendor/cmake/tinyxml2_vendorConfig.cmake:41 (include)
  /opt/ros/foxy/share/pluginlib/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/foxy/share/pluginlib/cmake/pluginlibConfig.cmake:41 (include)
  CMakeLists.txt:15 (find_package)

---
Failed   <<< hardware_interface [1.25s, exited with code 1]

Summary: 0 packages finished [1.42s]
  1 package failed: hardware_interface
  1 package had stderr output: hardware_interface
```

I'm using Foxy and I have installed the dependencies listed in the ros2_control_demo's readme, i.e. ros-foxy-realtime-tools ros-foxy-xacro ros-foxy-angles. I also thought it might be because TinyXML might be missing; however, libtinyxml2 is installed and up-to-date. 

Any help in resolving this issue would be greatly appreciated.

Best regards,

Frazer

destogl, 2020-12-17:
Probably a stupid question, but did you sourced your workspace before compiling?

P.S. I did multiple clean builds recently I never got this error, therefore I am a bit confused...

DrFKNoble, 2020-12-17:
@destogl I opened a fresh terminal and ran the following commands:
```console
source /opt/ros/foxy/setup.bash
cd ~/ros2_control
colcon build
```

bmagyar, 2020-12-18:
Our CI check against latest foxy binaries as well as latest foxy from source on Ubuntu. Is it possible you are running something else?

destogl, 2020-12-19:
@DrFKNoble it seems you didn't set up ros workspace. If you are trying to build inside a package you will get error since behaviour is not clear.
You should create proper ROS workspace, clone repos into src folder and than go to the workspace root and compile everything.
Using cmd something like:

```
mkdir -p ~/workspce/ros2_foxy_ws/src
cd ~/workspce/ros2_foxy_ws/src
git clone <ros2_control, ros2_controllers, ros2_control_demos>
cd ~/workspce/ros2_foxy_ws
source /opt/ros/foxy/setup.bash
colcon build
```

closed 2021-02-16 by bmagyar