ros-controls/ros2_control #282: Lack of access to former release repos in ros2-gbp org

bmagyar, 2020-12-23:
The dashing release was made using the release repos in `ros2-gbp` but I couldn't simply add the new releases on top of them as I have no access there. Opted for recreating the release repos as a personal repo but ideally I'd like to use

* https://github.com/ros2-gbp/ros2_control-release and
* https://github.com/ros2-gbp/ros2_controllers-release
for making releases in the future.

Who is there to ask for admission to `ros2-gbp`? Do you guys have any pointers @Karsten1987 @chapulina @ahcorde @clalancette ?

nuclearsandwich, 2020-12-23:
@bmagyar I've sent you an invitation to maintain both release repositories. Please let me know if you have any issues and feel free to close this issue once you have access.

bmagyar, 2020-12-23:
Grand! Thanks @nuclearsandwich , I now have access, upcoming releases will happen using those repos :+1:

closed 2020-12-23 by bmagyar