ros-controls/ros2_control #286: joint_limits_interface

ar0usel, 2020-12-30:
I'm trying to build `gazebo_ros2_control` package, and it's depends on `joint_limits_interface` package. While building `joint_limits_interface` occurred 
```
fatal error: hardware_interface/joint_handle.hpp: No such file or directory
 #include <hardware_interface/joint_handle.hpp>
```
Can you advise version which capable with `gazebo_ros2_control`? 

My setup: Ubuntu 18.04, ros2-foxy

ar0usel, 2020-12-30:
#https://github.com/ros-simulation/gazebo_ros2_control/issues/42

bmagyar, 2020-12-30:
Wait for the release my friend

closed 2020-12-30 by bmagyar

bmagyar, 2021-01-06:
We haven't made up our minds about what should happen to this package, most likely it will be merged into `hardware_interface`.