ros-controls/ros2_control #290: catkin_lint on the CI

bmagyar, 2021-01-07:
If we had catkin_lint as part of our linting workflow on the CI, it'd highlight things that derail the release farm but may not be otherwise highlighted with normal CI jobs or local compilations.

destogl, 2021-01-08:
Why catkin? Isn't this ROS1.

bmagyar, 2021-01-08:
I added this to the meeting agenda so I can explain it further but let me do it here then:

`catkin_lint` has the neat feature of highlighting cmake find_package() calls to ROS packages that weren't declared in the package.xml and vice-versa. It also highlights a couple of common mistakes that may not reveal themselves on local, or CI builds, but only in truly isolated builds which for us currently only happens on the buildfarm which is quite late to detect these issues. 

We have a lot of linting already in place for stuff that is far less crucial, I suggest we poke around the ament linting tools for something that can help us.

Perhaps @claireyywang is the right person to inform us on whether there is such a tool for ament?

bmagyar, 2021-04-02:
@Karsten1987 has this been raised before? This would be a very useful linter, checking ament dependencies vs cmake find_package calls.

Karsten1987, 2021-04-02:
I am pretty certain this won't be part of the roadmap in the near future.
My bet would be that this had to be a community contribution if so.

closed 2022-06-28 by bmagyar