ros-controls/ros2_control #291: Redesign transmission XML format

bmagyar, 2021-01-08:
Before `transmission_interface` can be released, we should carefully think about how we should design getting parameters through to transmissions within a SystemInterface implementation and an ActuatorInterface implementation. 

I am under the impression that transmission_parser is not even required anymore. What is your take on this @destogl ?

destogl, 2021-01-09:
At the current point, we don't need this since users can use the transmissions directly in their implementations. Transmissions don’t have to be loaded using pluginlib. 

Transmission parameters could be added as user-defined parameters for the hardware.

This is the easiest and most promising way. Let’s keep it simple for now. We can always make this more complex in the future if there is a plausible reason to do so.

bmagyar, 2021-01-09:
Transmissions still have parameters that need to be specified as well as they may not use the same type of transmissions for mapping things, hence pluginlib could be required - that being said this is a SystemInterface implementation detail.

bmagyar, 2021-01-11:
```
  <ros2_control name="LeftArmDriver" type="system">
    <hardware>
      <plugin>ros2_control_demo_hardware/MyRobotArmDriver</plugin>
      <param name="can_bus">2</param>
    </hardware>
    ...
    <joint name="arm_left_5_joint">
      ...
    </joint>
    <joint name="arm_left_6_joint">
      ...
    </joint>
    <joint name="arm_left_7_joint">
      ...
    </joint>
    
    <transmission name="arm_left_5_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <actuator name="arm_left_5_motor" >
        <mechanicalReduction>13.37</mechanicalReduction>
      </actuator>
      <joint name="arm_left_5_joint">
        <offset>0.01</offset>
      </joint>
    </transmission>
    
    <transmission name="arm_left_wrist_trans">
    <type>transmission_interface/DifferentialTransmission</type>
      <actuator name="arm_left_6_motor">
        <role>actuator1</role>       
        <mechanicalReduction>1.0</mechanicalReduction>
      </actuator>
     <actuator name="arm_left_7_motor">
        <role>actuator2</role>
        <mechanicalReduction>1.0</mechanicalReduction>
      </actuator>
      <joint name="arm_left_6_joint">
        <role>joint1</role>
        <mechanicalReduction>0.5</mechanicalReduction>
      </joint>
      <joint name="arm_left_7_joint">
        <role>joint2</role>
        <mechanicalReduction>0.3</mechanicalReduction>
      </joint>
    </transmission>
  </ros2_control>
  ```

@jordan-palacios if you have other input, we'd appreciate it

jordan-palacios, 2021-01-12:
This looks good an easily extendable.

As @bmagyar says, it's specially important to support generic parameters. This will allow the development of more complicated transmission types.

What I'm not really sure yet if transmissions should be plugins.

> How should we go about configuring which interfaces to load for joint/actuator?

A parameter per joint/actuator with the list of interfaces to configure?

destogl, 2021-01-13:
> What I'm not really sure yet if transmissions should be plugins.

Me too. The users will anyway have them "hard-coded" into their system/actuator. I am not sure how plugins should provide any advantage.

bmagyar, 2021-01-13:
@guru-florida @ahcorde

bmagyar, 2021-05-05:
https://github.com/ros-controls/roadmap/pull/44

closed 2021-05-05 by bmagyar