ros-controls/ros2_control #293: Get ros2_control into REP 2005

destogl, 2021-01-13:
[REP 2005](https://www.ros.org/reps/rep-2005.html)

### Reasoning
- ros2_conrol is an integral part of ROS2 and will be used by many people
- companies in ROS2 TSC can get their work counted

### Current state w.r.t. the criteria
- [x] open source 
- [x] connected to ROS2 project
- [x] broadly useable
- [x] actively maintained and following accepted software development processes, including code review and testing
- [ ] evidently used by a nontrivial part of the community
- [ ] are maintained by an organization or at least two individuals
- [ ] Quality Level 3 as defined in [REP 2004](https://www.ros.org/reps/rep-2004.html)

### Open issues
1. The community will build on its own in the next few months. (practically not the issue)
2. Currently there is only one official maintainer (?) @bmagyar is that right? Or @Karsten1987 is also counted for?
3. We have to clarify the QL. I think we are above 3 when everything counted. We should open a project to achieve [QL2](https://www.ros.org/reps/rep-2004.html#id49).

bmagyar, 2021-01-13:
1. We can prove this part met based on ROS1 ros_control.
2. Both Karsten and myself are declared as leads of the Control Working Group, I think this should be enough. Also our packages have even more ppl listed as maintainers and we can show a good range of people providing reviews.
3. We could probably declare Quality 3 already and open a project for QL2

bmagyar, 2021-01-15:
https://github.com/ros-controls/ros2_control/issues/295