ros-controls/ros2_control #294: Spawner and unspawner

bmagyar, 2021-01-13:
Spawner: simple python nodes that load, configure and start a controller on startup, and stops, unloads on takedown.

Unspawner: the opposite of spawner

Typical use-case with robots that come with ros2_control by default on startup software deployed by manufacturer of a mobile manipulator, my launch file would:
* unspawn arm_joint_trajectory_controller
* spawn my_awesome_force_arm_controller

When my launch file is CTRL+C-d, robot is back to normal operation.

bmagyar, 2021-01-13:
FYI @JafarAbdi

JafarAbdi, 2021-01-14:
Thanks! @bmagyar, let me know if there's something I can help with, for now as a workaround, I add the following action to the launch description

```python
ExecuteProcess(cmd=['ros2 control load_configure_controller panda_arm_controller'], shell=True, output='screen', on_exit=[ExecuteProcess(cmd=['ros2 control switch_controllers --start-controllers panda_arm_controller'], shell=True, output='screen')])
```

v-lopez, 2021-01-26:
How would you guys feel if the spawner also loaded an optionally provided yaml file with the controller parameters?

Because right now I have to do:

1. start controller_manager with the proper `my_controller_name.type` param
2. ros2 control load_controller **my_controller_name** (creates the controller Node)
3. load **my_controller_name** parameters into the namespace
4. ros2 control configure_start_controller

closed 2021-03-28 by bmagyar