ros-controls/ros2_control #298: problem with resource manager when validate storage

yimenghou, 2021-01-16:
I guess the content in validate_storage() may not finished. There is a typo in hardware_interface/resource_manager.cpp line 377, where 'missing_command_keys' should emplace back resources instead of 'missing_state_keys'. Which may cause loading error. Besides that, hardware sensors are not validated here yet.

bmagyar, 2021-01-17:
We aren't currently extending this section but would be happy to see if you
can produce a test case that fails or just doesn't work as intended. We
could use that as a test and have a task for someone to fix.

bmagyar, 2021-02-03:
@Karsten1987 ?

destogl, 2021-02-03:
> I guess the content in validate_storage() may not finished. There is a typo in hardware_interface/resource_manager.cpp line 377, where 'missing_command_keys' should emplace back resources instead of 'missing_state_keys'. Which may cause loading error.

This seems correct. We have to work on a test case for this. We need to produce an URDF to provoke this error.

> Besides that, hardware sensors are not validated here yet.

They should be [here](https://github.com/ros-controls/ros2_control/blob/40475642201e71d37ca4e622fad48ba0d6b4393f/hardware_interface/src/resource_manager.cpp#L381).

destogl, 2021-02-03:
@Karsten1987 I work on this part for the framework this week anyway. I can than notify what is result of my tests.

Karsten1987, 2021-02-03:
@yimenghou thanks for bringing this up. Line 377 is clearly a typo. I'll come up with a PR to fix this if you haven't started on it yet.

> Besides that, hardware sensors are not validated here yet.

What does that mean? I do believe that the validation for the sensors should be sufficient in here: https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/src/resource_manager.cpp#L381-L387

Karsten1987, 2021-02-03:
> I work on this part for the framework this week anyway. I can than notify what is result of my tests.

@destogl I consider this a clear bug and would open a separate PR for it.

yimenghou, 2021-02-03:
Yes, hey are validated. I didn't see it right. Thanks!

> What does that mean? I do believe that the validation for the sensors should be sufficient in here: https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/src/resource_manager.cpp#L381-L387

PR #317 (2021-02-05): correct hardware interface validation in resource manager.
fixes #298 

cc @yimenghou
closed 2021-02-05 by bmagyar