ros-controls/ros2_control #305: Extending the CommandInterface

olivier-stasse, 2021-01-24:
Thanks again for the work realized.

I am trying to write a rrbot_system_quadruped with an extended CommandInterface as discussed during the meetings.
(A preliminary version is [here](https://github.com/olivier-stasse/ros2_control_demos) )
Instead of having a double for the StateInterface or the Command Interface the goal is to have:
https://github.com/olivier-stasse/ros2_control_demos/blob/59ecf64ab6b4acc36407fa8292a248b683d47514/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_quadruped.hpp#L38
`struct PosVelEffortGains
{
  double position;
  double velocity;
  double effort;
  double Kp;
  double Kd;
}`

The way the CommandInterface is written now I do not see how to implement this.

Should I reimplement a `ReadWriteHandle` class with another type. Or would it be better to have a `ReadWriteHandle` with a new template parameter ?

I probably did not get the architecture. Any pointer, suggestion would be deeply appreciated.

olivier-stasse, 2021-01-25:
Or should I keep ```hardware_interface::HW_IF_POSITION``` ```HW_IF_VELOCITY``` ```HF_IF_EFFORT``` and add inside ```RRBotSystemQuadrupedHardware``` specific local name such as ```HW_IF_GAIN_KP``` and ```HW_IF_GAIN_KD``` ?

destogl, 2021-01-26:
Hi @olivier-stasse,

Since you have a structure with all-double, there is no reason to extend the `CommandInterface` or `StateInterface`. IMHO this would lead to incompatibility with controllers and make the framework unnecessarily complex.

In your specific case (when having many double values) I propose using of a separate interface for each value. So you will export for each joint following (state and command) interfaces:

```
hardware_interface::StateInterface(info_.joints[i].name, "position", &hw_states_pos_[i]));
hardware_interface::StateInterface(info_.joints[i].name, "velocity", &hw_states_vel_[i]));
hardware_interface::StateInterface(info_.joints[i].name, "effort", &hw_states_eff_[i]));
hardware_interface::StateInterface(info_.joints[i].name, "Kp", &hw_states_kp_[i]));
hardware_interface::StateInterface(info_.joints[i].name, "Kd", &hw_states_kd_[i]));
...
```

Those will be then automatically matched with your controller requesting the same names. Alternatively, to keep the structure cleaner in your hardware driver and the specific controller, you could do the following:
```
std::vector<PosVelEffortGains> hw_states_;  // in .hpp file

// in export_state_interfaces (.cpp)
hardware_interface::StateInterface(info_.joints[i].name, "position", &hw_states_[i]).position);
hardware_interface::StateInterface(info_.joints[i].name, "velocity", &hw_states_[i].velocity);
hardware_interface::StateInterface(info_.joints[i].name, "effort", &hw_states_[i].effort);
hardware_interface::StateInterface(info_.joints[i].name, "Kp", &hw_states_[i].Kp);
hardware_interface::StateInterface(info_.joints[i].name, "Kd", &hw_states_[i].Kd);
...
```
Would this work for you?

> Or should I keep hardware_interface::HW_IF_POSITION HW_IF_VELOCITY HF_IF_EFFORT and add inside RRBotSystemQuadrupedHardware specific local values for the interface such as HW_IF_GAIN_KP and HW_IF_GAIN_KD ?

You can define additional local types if you like or hard-code them if they are not used often.

olivier-stasse, 2021-01-26:
Thanks @destogl ! This is currently the direction I am taking.

closed 2021-01-26 by olivier-stasse