ros-controls/ros2_control #306: Some cli commands return '1' even through it's successfully executed

ShotaAk, 2021-01-26:
Some `ros2 control` commands return exit-status '1' even through it's successfully executed.

# Environments
- OS: Ubuntu 18.04
- Software Version: latest of master, commit: fb804313a2a7d3992ef58034a369e8558f7f8c57

# To Reproduce

```sh
# Terminal 1
$ ros2 launch ros2_control_demo_robot rrbot_system_position_only.launch.py

# Terminal 2
$ ros2 control list_hardware_interfaces
$ echo $?
0

$ ros2 control load_start_controller joint_state_controller
$ echo $?
1

$ ros2 control load_configure_controller forward_command_controller_position
$ echo $?
1

$ ros2 control switch_controllers --start-controllers forward_command_controller_position 
$ echo $?
1
```

# Expected behavior

The commands return the exit-status '0'.

destogl, 2021-01-26:
The issue is that some verbs are returning the results from service_caller (`future.result()`). See [here](https://github.com/ros-controls/ros2_control/blob/master/ros2controlcli/ros2controlcli/verb/load_start_controller.py#L33).

What is the proper thing to do? Inverse the return? I would guess so ([see](https://docs.python.org/3/library/asyncio-future.html#asyncio.Future.result)).

ShotaAk, 2021-01-26:
> Inverse the return? 

I agree with your suggestion.:+1:

ShotaAk, 2021-01-26:
I will create a PR.

PR #308 (2021-01-28): Inverse the response of cli commands to return correct exit-status.
Fix #306 .
closed 2021-01-28 by bmagyar