ros-controls/ros2_control #311: I know this is a larger change. but, would this name be better?

destogl, 2021-01-26:
I know this is a larger change. but, would this name be better?

```suggestion
  std::vector<JointInterface> joint_position_;
```

__Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/303#discussion_r564325294__

bmagyar, 2021-01-28:
so the reason I wanted to stick with handles is that essentially `transmission_interface` is a library that is an optional add-on for ros_control. Similarly to how they were optional in ROS1 (your `RobotHW` didn't need to implement them) but here I wanted to step away from using the same handles, hence the choice to use `JointHandle` and `ActuatorHandle` which are now unique to `transmission_interface`.

If we were to rename `JointHandle` to `JointInterface`, what'd that mean mean, how would that related to `StateInterface` or `CommandInterface`? In the `hardware_interface` scope we only use joint interfaces, no actuator ones. Also, would we also rename `ActuatorHandle` to `ActuatorInterface`? We already have a class like this in `hardware_interface` which serves an entirely different purpose.

destogl, 2021-02-02:
@bmagyar this is very plausible. Let document this somewhere. Than I agree to close this.

destogl, 2021-02-03:
Since this is doc issue, and there are many, many more I will close this here.

closed 2021-02-03 by destogl