ros-controls/ros2_control #314: Post-Foxy breaking changes

bmagyar, 2021-02-03:
Let's collect the known API/ABI breaking changes we'd like to see in a release coming after Foxy.

* [ ] Move constants out of class & refactor to avoid lengthy casting in [controller_manager constants issue](https://github.com/ros-controls/ros2_control/pull/312/files#r568395744)
* [ ] Update actions: [ros2_controllers#111](https://github.com/ros-controls/ros2_controllers/pull/111)

tylerjw, 2021-09-08:
> Move constants out of class & refactor to avoid lengthy casting in controller_manager constants issue

The lengthy cast seems to be gone: https://github.com/ros-controls/ros2_control/blob/7a0d5169c6a0f4aa8381b0e57de22ee28e6172bf/controller_manager/include/controller_manager/controller_manager.hpp#L126

tylerjw, 2021-09-09:
I rebased 111 on ros2_controllers here: https://github.com/ros-controls/ros2_controllers/pull/237  It needs the CI job approved before it can run because this is my first time opening a PR here.

bmagyar, 2021-09-30:
Thanks for everyone who helped out, releases are now on the way!

closed 2021-09-30 by bmagyar