ros-controls/ros2_control #318: Non-Joint Command interfaces

destogl, 2021-02-03:
Currently, we defined two types of interfaces: related to a joint and related to a sensor. What if there is a command interface needed that is not related to any joint? An example of this is that the robot has voltage output on its flange. Ideally, this interface is also controlled by a controller, e.g. Forward Command Controller can write to an arbitrary interface. To realize this the following URDF-snippet should be defined:

```
  <ros2_control name="GenericSystem2dof" type="system">
    <hardware>
      <plugin>fake_components/GenericSystem</plugin>
    </hardware>
    <joint name="joint1">
      <command_interface name="position"/>
      <command_interface name="velocity"/>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <param name="initial_position">0.0</param>
      <param name="initial_velocity">0.0</param>
    </joint>
    <joint name="joint2">
      <command_interface name="position"/>
      <command_interface name="velocity"/>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <param name="initial_position">0.0</param>
      <param name="initial_velocity">0.0</param>
    </joint>
    <joint name="voltage_output">
      <command_interface name="voltage"/>
      <state_interface name="voltage"/>
      <param name="initial_voltage">0.0</param>
    </joint>
  </ros2_control>
```

What disturbed me here is the use of `joint`-tag with `voltage_output`. This output is not a joint but rather a GPIO-type interface. Also, ideally, in the future, we will map joints from the kinematics part to joints defined under the `ros2_control` tag, e.g., for realizing joint limits. This could be a potential obstacle there.

Should we define an additional type of read/write interfaces that are not joints? Proposal: let's call them `gpio`.

@Karsten1987, @bmagyar, @olivier-stasse What's your opinion on this?

@livanov93 you had a similar question/issue with UR driver? How would you like to have it realized?

olivier-stasse, 2021-02-03:
IMHO `gpio` is better than `joint`.

destogl, 2021-02-04:
> IMHO `gpio` is better than `joint`.

I agree. Although the functionality is the same, this semantic increases clarity.

gavanderhoorn, 2021-02-04:
what about making things a little more generic?

It seems command interfaces are sinks and state interfaces are sources.

Might be a nice concept to make explicit in the codebase.

destogl, 2021-02-05:
> what about making things a little more generic?
> 
> It seems command interfaces are sinks and state interfaces are sources.

This is a very good idea. Nevertheless, I don't see an issue with the interfaces, but rather how they are structured.
An example would be a robot that has a controllable voltage output at its flange (see #323). It ros2_control part of the URDF is as follows:

```
 <ros2_control name="GenericSystem2dof" type="system">
    <hardware>
      <plugin>fake_components/GenericSystem</plugin>
    </hardware>
    <joint name="joint1">
      <command_interface name="position"/>
      <command_interface name="velocity"/>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <param name="initial_position">0.0</param>
      <param name="initial_velocity">0.0</param>
    </joint>
    <joint name="joint2">
      <command_interface name="position"/>
      <command_interface name="velocity"/>
      <state_interface name="position"/>
      <state_interface name="velocity"/>
      <param name="initial_position">0.0</param>
      <param name="initial_velocity">0.0</param>
    </joint>
    <joint name="voltage_output">
      <command_interface name="voltage"/>
      <state_interface name="voltage"/>
      <param name="initial_voltage">0.5</param>
    </joint>
  </ros2_control>
```

It would probably make sense to change the last 6 lines to:

```
    <gpio name="voltage_output">
      <command_interface name="voltage"/>
      <state_interface name="voltage"/>
      <param name="initial_voltage">0.5</param>
    </gpio>
  </ros2_control>
```

I hope, now it's clearer what is issue here.

livanov93, 2021-02-07:
At [component_parser](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/src/component_parser.cpp) I see that all the tags are hardcoded. It would generally be handy to be able to define new tag within ros2_control xml tag. This way the user could define naming convention for each GPIO or robot subsystem. 
Of course, this does not engage the problem we currently have - only values of type double can be exposed/set directly from/to the lower level hardware (system, sensor, actuator)

To make things less complicated at first it would be nice to add something like `gpio` tag in xml definition. `gpio` tag is fine, maybe something like `generic_robot_component` tag is something that sounds nice too.

AndyZe, 2021-02-10:
Agree that something like this shouldn't be labeled a `joint`. We talked about why this should be part of ros2_control -- there may be a robot controller, which the hardware interface has access to. It might not be available to any other ROS nodes. So it makes sense (to me) to have this in ros2_control.

AndyZe, 2021-02-10:
I think some kind of alias called `gpio` for readability's sake is nice!  :+1:

ravnicas, 2021-04-16:
`gpio` would describe some in- or ourput of a controller or external sensor. Would an additional alias, f.e. `flag` be also an option for boolean interfaces of OPCua hardware for example?

closed 2022-07-17 by bmagyar