ros-controls/ros2_control #320: Feature proposal helper function for loading controllers via launch.py file

v-lopez, 2021-02-05:
Just testing the waters with some code to facilitate writing launch files for loading controllers. 
Could I get some feedback as to whether I should try to make it into ros2_control or just keep it for ourselves?

Our use case is that our robots don't always run the same controllers. Even may run controllers that some of our customers build and we don't have the yaml parameters for them when we ship the robot.

Our solution in ROS1 is to load the controller parameters and type AFTER ros_control is running, and switch to the new controllers.

A similar approach would be desirable for us. But launch.py files in ROS2 are a bit harder to write, so I propose adding a python module that can be included in launch that simplifies loading params and spawning the controller.

```python
def generate_load_controller_launch_description(controller_name, controller_type, pkg_name, controller_params_file):

    pkg_path = get_package_share_directory(pkg_name)
    param_file_path = os.path.join(pkg_path, controller_params_file)

    spawner = Node(package='controller_manager', executable='spawner.py',
                    arguments=[controller_name,
                        '--controller-type', controller_type,
                        '--controller-manager', LaunchConfiguration('controller_manager_name'),
                        '--param-file', param_file_path],
                        output='screen')

    return LaunchDescription([
        DeclareLaunchArgument(
            'controller_manager_name', default_value='controller_manager',
            description='Controller manager node name'
        ),
        spawner,
    ])
```

And later, to load the `mobile_base_controller` of type `diff_drive_controller/DiffDriveController` you just have to write a launch file with:
```
def generate_launch_description():

    return generate_load_controller_launch_description(controller_name='mobile_base_controller',
                                                       controller_type='diff_drive_controller/DiffDriveController',
                                                       pkg_name='pmb2_controller_configuration',
                                                       controller_params_file=os.path.join('config', 'default_controllers.yaml'))
```

Many improvements can be made, some arguments (like `controller_type` or `controller_params_file`) could be optional, but just would like to get some feedback as to whether I should add this to `controller_manager` or just keep it for us on one of our repos.

destogl, 2021-02-12:
@v-lopez why not. The only question is where to place such code snippets.

v-lopez, 2021-02-15:
I'd say `controller_manager` or a new `ros2_control_launch` in this repository.

v-lopez, 2021-02-16:
I added them on `controller_manager` on #310 

We can move to a separate package if you see fit

destogl, 2021-02-16:
For me is fine to be there

commit bea91563 (2021-03-29): Add launch_utils

fixes #320
closed 2021-04-02 by bmagyar