ros-controls/ros2_control #341: ros2control CLI simplification

bmagyar, 2021-03-10:
@Karsten1987 could you please re-iterate your proposal here and one of us could get around to implement it?

Karsten1987, 2021-03-10:
The current set of CLI options of `ros2 control` are a bit overwhelming and not really user-friendly. Especially the load and state changes wrt the individual controllers are a bit cumbersome to deal with:

```
  configure_controller         Configure a controller in a controller manager
  configure_start_controller   Configure and Start a controller in a controller manager
  [..]
  load_configure_controller    Load and Configure a controller in a controller manager
  load_controller              Load a controller in a controller manager
  load_start_controller        Load, Configure and Start a controller in a controller manager
  [..]
  unload_controller            Unload a controller in a controller manager
```

I propose to cluster these services a bit more tightly:
```
ros2 control load_controller <controller_name> <controller_type> --param-file </path/to/param>
ros2 control unload_controller <controller_name>

ros2 control set_controller_state [configure | start | stop | shutdown] <controller_name>
```

Alternatively, for the last command I could see a second top-level verb `ros2 controller`:
```
ros2 controller configure <controller_name>
ros2 controller start <controller_name>
ros2 controller stop <controller_name>
```

The first set of commands for loading and unloading essentially reflects the changes in #310

destogl, 2021-04-02:
@Karsten1987 this should be done in #349? Isn't it? Please reopen if not.

closed 2021-04-02 by destogl