ros-controls/ros2_control #345: Question regarding the releasing of controller's claimed interfaces

ssumoo, 2021-03-13:
Hi, I have a question regarding the release of the controller's claimed interfaces. As I am new to the codebase, I am not sure whether or not this is indeed an issue or a deliberate design choice that I am not yet aware of.

In short, the question is: _why does neither of the following happen?:_
1. _The controller manager calls the controller's `release_interfaces` in its `stop_controllers` function_
2. _The controller calls its own `release_interfaces` in its `deactivate` function_

The reasoning behind this question is that the resource manager of the controller manager claims the required interfaces for the controllers that are being started in the `start_controllers` function, which makes sense in a way that the controller should have the interfaces before moving into the active state.

Although I'm aware of the fact that the resource manager binds its `release_command_interface` as the deleter of the loaned interfaces when they are being given to the controller, it seems to be the case that the controller needs to be destructed to release its claimed interfaces, instead of simply releasing the claimed interfaces when it is stopped.

I have checked the ros2_controllers repo as well, and it seems that the trajectory controller is releasing its claimed interfaces when it is deactivated, but not other controllers.

This means that switching between controllers which have conflicting joints with a single service call will not be possible, because the controllers being stopped will not release their interfaces, while those being started will try to claim the interfaces, and fail.

Thank you for reading, if it is indeed an intended design, I'd like to know the reasoning behind it, along with an easy way to switch between controllers in my use cases (and then close this issue), or else I'll be happy to contribute as required.

bmagyar, 2021-03-15:
Very good attention to detail! Indeed the bound-deleter is more of a last resort safety to ensure we are not leaking interfaces. First of all, please ensure that you are running with latest release/master, not something old. 

To drive your point home, you could bring up rrbot from the demos repo and load a `joint_trajectory_controller` and a `position_controllers/JointGroupPositionController` and try swapping between them. Feel free to post results and steps to reproduce here so that others can confirm too.

I think this is something that could probably deserve a sort of integration test for each controller. You could declare two controllers of the same type, same parameter but different names. Load them both, start one, deactivate, try starting the other.

Karsten1987, 2021-03-15:
@ssumoo thanks for opening this ticket. As I believe your encountered behavior is the same as being addressed in #343, I'd close this ticket here and would very strongly invite you to participate on the discussion over there.

If you feel that's fundamentally different, please re-open your ticket.

closed 2021-03-15 by Karsten1987

ssumoo, 2021-03-15:
@bmagyar Thank you for the feedback, actually, I did the exact experiments you mentioned, where I tried to switch between two different forward command controller, each of which claims the same joints.

@bmagyar @Karsten1987 Thank you for pointing the MR out, I'm happy that it matches perfectly with what I patched locally on the repo, I'll be waiting to pull the changes then :D

Karsten1987, 2021-03-15:
@ssumoo it's great that you have already an idea on how to test. I am sure any contributions to #343 are welcome.