ros-controls/ros2_control #352: load controllers with a namespaced controller manager by control cli

yimenghou, 2021-03-29:
Hi,

Could I use the ros2 control cli with namespaced CM?

For example, lets say, my robot has a namespace with `bot1`. I want to bringup a CM with 'bot1' as namespace in the launch file.
After launching, I should be given a ros2 node "bot1.controller_manager". Then I want to `use ros2 control load_start_controller example_controller` to load a controller. The cli reports "Could not contact service /controller_manager/load_and_start_controller". Obviously It cannot find this CM node with namespace. But `ros2 service list` do can find that node. Is there any way to do this? Thanks.

yimenghou, 2021-03-29:
I think i find the answer. The load_controller has a -c option to specify the complete node (with namespace)

closed 2021-03-29 by yimenghou

arjun-kg, 2021-04-02:
Hi, how were you able to change the namespace of the controller_manager which is created by the plugin?

yimenghou, 2021-04-02:
You can add namespace in the yaml config file at the first level indentation.