ros-controls/ros2_control #355: use format strings for ros2controlcli

Karsten1987, 2021-03-29:
rather than using `.format()`, we can use `f'...'` in Python3.6

see https://www.python.org/dev/peps/pep-0498/

NovusEdge, 2021-03-30:
I would like to take this issue up if it's still open.

bmagyar, 2021-03-30:
I'll put a few examples of what should be replaced and vaguely how, they may not always work, be creative ;) 

examples:
* `node.get_logger().debug('requester: making request: %r\n' % request)` ->  `node.get_logger().debug(f'requester: making request: {request}\n')`
*  `return service_caller(node, '{}/switch_controller'.format(controller_manager_name), SwitchController, request)` -> `return service_caller(node, f'{controller_manager_name}/switch_controller', SwitchController, request)`
* `node.get_logger().info('Loaded ' + param_file + ' into ' + controller_name)` -> `node.get_logger().info(f'Loaded {param_file} into {controller_name}')`

bmagyar, 2021-03-30:
@NovusEdge I've assigned you, please tag this issue in the PR you open for reference

PR #358 (2021-04-29): [python] Update files in ros2controlcli to use format strings
Fixes #355 

@bmagyar 
Please be sure to review the changes and let me know if something is wrong; or if more changes are to be made.
closed 2021-04-29 by bmagyar