ros-controls/ros2_control #356: pep257 differences between the ros-tooling and ros-industrial CI

bmagyar, 2021-03-30:
By @Karsten1987 
> I do believe there's a fine difference in the pep257 between the ros-tooling and ros-industrial CI. The source build (ros-tooling) seems to use a version which expects a semicolon ; after a doc section, whereas the binary jobs (ros-industrial) don't.
Not sure what to do here.

https://github.com/ros-controls/ros2_control/pull/349

https://github.com/ros-controls/ros2_control/pull/349/commits/0df21e18a8f8c7e5941034344a5222c8e29cf19c

bmagyar, 2021-03-30:
I think we should simply remove the non-tests from https://github.com/ros-controls/ros2_control/tree/master/ros2controlcli/test and let the linter workflows do their job. Those individual scripts are prune to be flaky depending on ones installed python package versions which is I believe what we are seeing with the industrial CI.

destogl, 2021-04-02:
why not to move this into pre-commits. See here: https://www.youtube.com/watch?v=tOlnfrctpSY

Karsten1987, 2021-04-02:
what does this mean to remove the non-tests? There aren't any inbuilt
linters for python. THese files are unfortunately copied in every package
which has python code, e.g:
https://github.com/ros2/demos/tree/master/demo_nodes_py/test

bmagyar, 2021-04-04:
Yes and that is a problem. We are carrying dead weight. Don't want to follow a bad example. Precommit for instance is able to reliably reproduce linting work both locally editing (some) and CI validating without the flakiness of dependig and on which pytest or other tools are installed in the environment.

Karsten1987, 2021-04-06:
sure, I am happy to have a better solution for this - as long as it's kind of transparently incorporated so that I can be able to test locally and have CI do the exact same thing as well.

PR #1626 (2024-07-20): Remove noqa
We don't use ament_pep257 any more, but use pydocstyle from the pre-commit config.

Closes #356
PR #1628 (2024-07-20): Remove noqa (backport #1626)
We don't use ament_pep257 any more, but use pydocstyle from the pre-commit config.

Closes #356 <hr>This is an automatic backport of pull request #1626 done by [Mergify](https://mergify.com).
PR #1629 (2024-07-20): Remove noqa (backport #1626)
We don't use ament_pep257 any more, but use pydocstyle from the pre-commit config.

Closes #356 <hr>This is an automatic backport of pull request #1626 done by [Mergify](https://mergify.com).
closed 2024-07-20 by christophfroehlich