ros-controls/ros2_control #357: Documentation and/or tests for ros2controlcli

bmagyar, 2021-03-30:
With https://github.com/ros-controls/ros2_control/pull/349 we've introduced a new verb, `ros2 control set_state ...` and added an optional  parameter to `ros2 control load ...` to support the same logic, advancing the state of the controller to a desired state with supplying `--state ...`  to the load command. With this, the previous "multitool" CLI verbs are considered deprecated  `configure_start_controller`, `load_configure_controller` and `load_start_controller`.

We currently don't have documentation or functional tests for our CLI verbs.

For tests I don't think we need more than simplistic integration tests that verify that they exist and work as expected, regardless of back-end changes.
 
For documentation, I will add an initial version to the package and extend the control.ros.org scripts to pick it up and compile it into the page as well.

Opinions? @destogl @Karsten1987 @AndyZe ?

christophfroehlich, 2023-02-21:
I had a look on the documentation of the cli while writing the other PR yesterday, @bmagyar do you think there is something still missing?

Regarding tests: 

- I'd like to write some of the missing tests for the cli, do you have a starting point for me how this could look like?
- https://github.com/ros-controls/ros2_control/blob/master/ros2controlcli/test/test_view_controller_chains.py is not executed by colcon in my environement, only if I add `--python-testing pytest` or add `__init__.py` to the test folder. Is there something wrong with my installation or is something missing for the test configuration?

christophfroehlich, 2025-04-03:
Update regarding tests: We have some of them integrated into ros2_control_demo tests, but this is not covering all verbs.

christophfroehlich, 2026-03-17:
Sure. I suggest that you first get familiar with the cli verbs we have, how the status of the documentation is and which verbs are tested where, for example in the ros2_control_demos repository.

ielson, 2026-04-15:
I'll do it! When you have some time, could you just kick off the CI again? I have just fixed something I missed in the package.xml, so now it should pass the tests @christophfroehlich