ros-controls/ros2_control #383: ControllerManager::load_controller should return nullptr when ControllerInterface::init returns return_type::ERROR

AugusteBourgois, 2021-04-16:
According to [LoadController.srv](https://github.com/ros-controls/ros2_control/blob/master/controller_manager_msgs/srv/LoadController.srv), I believe that the result of the `ControllerInterface::init` function should be taken into account, which is not the case at the moment.

I propose that if `ControllerInterface::init` fails, then `nullptr` is returned, instead of a pointer towards a controller which initialization failed.

I can shoot a PR with a fix as soon as you agree (see [my branch](https://github.com/ros-controls/ros2_control/compare/master...AugusteBourgois:check_init_return_value))

destogl, 2021-04-17:
Great point! We will be glad to get PR from you?

AugusteBourgois, 2021-04-28:
The PR linking this issue has been merged. Can we close it ?

Karsten1987, 2021-04-28:
just fyi, for future PRs you can add a "fixes <link to issue>". Github will then close the ticket automatically once the PR is merged.

closed 2021-04-28 by Karsten1987