ros-controls/ros2_control #388: Changes to debug macros for Rolling

tylerjw, 2021-04-22:
When I try to build this repo on rolling I get this error:

```
/home/runner/work/moveit2/moveit2/.work/upstream_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:741:32: error: cannot convert ‘const std::__cxx11::basic_string<char>’ to ‘const char*’
    741 |     RCLCPP_DEBUG(get_logger(), cur_type);
        |                                ^~~~~~~~
        |                                |
        |                                const std::__cxx11::basic_string<char>
```

Here is a link to the CI job where you can see this error: https://github.com/ros-planning/moveit2/runs/2413992645

bmagyar, 2021-04-23:
could probably be fixed already by `cur_type.c_str()`, right? 

once I made an accidental release to rolling and once half the buildfarm was on fire we ended up removing it

tylerjw, 2021-04-23:
Testing it locally it looks like you can just replace the `RCLCPP_DEBUG` with `RCLCPP_DEBUG_STREAM` to fix this.  I'll make a small PR for that.

However, after making that change locally I'm now seeing this error:

```
--- stderr: transmission_interface                                                                               
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:67,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:51,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:14:
/opt/ros/rolling/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38: warning: ‘constexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [-Wdeprecated-declarations]
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/rolling/src/gtest_vendor/include/gtest/gtest-param-test.h:496:38: note: in definition of macro ‘INSTANTIATE_TEST_CASE_P’
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:51,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:14:
/opt/ros/rolling/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16: note: declared here
 1210 | constexpr bool InstantiateTestCase_P_IsDeprecated() { return true; }
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:67,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:51,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:14:
/opt/ros/rolling/src/gtest_vendor/include/gtest/gtest-param-test.h:496:73: warning: ‘constexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [-Wdeprecated-declarations]
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
      |                                                                         ^
/home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:170:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
  170 | INSTANTIATE_TEST_CASE_P(
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:62,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:51,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:14:
/opt/ros/rolling/src/gtest_vendor/include/gtest/internal/gtest-internal.h:1210:16: note: declared here
 1210 | constexpr bool InstantiateTestCase_P_IsDeprecated() { return true; }
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:67,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/internal/gmock-internal-utils.h:47,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock-actions.h:51,
                 from /opt/ros/rolling/src/gmock_vendor/include/gmock/gmock.h:59,
                 from /home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:14:
/opt/ros/rolling/src/gtest_vendor/include/gtest/gtest-param-test.h:496:73: warning: ‘constexpr bool testing::internal::InstantiateTestCase_P_IsDeprecated()’ is deprecated: INSTANTIATE_TEST_CASE_P is deprecated, please use INSTANTIATE_TEST_SUITE_P [-Wdeprecated-declarations]
  496 |   static_assert(::testing::internal::InstantiateTestCase_P_IsDeprecated(), \
      |                                                                         ^
/home/tyler/code/moveit2/ws_rolling/src/ros2_control/transmission_interface/test/simple_transmission_test.cpp:170:1: note: in expansion of macro ‘INSTANTIATE_TEST_CASE_P’
  170 | INSTANTIATE_TEST_CASE_P(
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:62,
[... cut ...]
/opt/ros/rolling/src/gtest_vendor/include/gtest/internal/gtest-param-util.h:382:27: error: static assertion failed: Empty arguments are not allowed.
  382 |   static_assert(sizeof(T) == 0, "Empty arguments are not allowed.");
      |                 ~~~~~~~~~~^~~~
make[2]: *** [CMakeFiles/test_simple_transmission.dir/build.make:63: CMakeFiles/test_simple_transmission.dir/test/simple_transmission_test.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:212: CMakeFiles/test_simple_transmission.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< transmission_interface [0.97s, exited with code 2]
```

tylerjw, 2021-04-23:
Closing this as it is now fixed.

closed 2021-04-23 by tylerjw