ros-controls/ros2_control #400: fail to build hardware_interfaces CMake Error at CMakeLists.txt:96 (find_package):

newcanopies, 2021-04-27:
Hi, 

Most of the packages in this repo build successfully, but not hardware_interfaces

```
Starting >>> hardware_interface
[Processing: hardware_interface]
--- stderr: hardware_interface
CMake Error at CMakeLists.txt:96 (find_package):
  By not providing "Findros2_control_test_assets.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "ros2_control_test_assets", but CMake did not find one.

  Could not find a package configuration file provided by
  "ros2_control_test_assets" with any of the following names:

    ros2_control_test_assetsConfig.cmake
    ros2_control_test_assets-config.cmake

  Add the installation prefix of "ros2_control_test_assets" to
  CMAKE_PREFIX_PATH or set "ros2_control_test_assets_DIR" to a directory
  containing one of the above files.  If "ros2_control_test_assets" provides
  a separate development package or SDK, be sure it has been installed.

---

nor  transmission_interface/

```
Starting >>> transmission_interface
--- stderr: transmission_interface
CMake Error at CMakeLists.txt:15 (find_package):
  By not providing "Findhardware_interface.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "hardware_interface", but CMake did not find one.

  Could not find a package configuration file provided by
  "hardware_interface" with any of the following names:

    hardware_interfaceConfig.cmake
    hardware_interface-config.cmake

  Add the installation prefix of "hardware_interface" to CMAKE_PREFIX_PATH or
  set "hardware_interface_DIR" to a directory containing one of the above
  files.  If "hardware_interface" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< transmission_interface [9.80s, exited with code 1]

Summary: 0 packages finished [10.1s]
  1 package failed: transmission_interface
  1 package had stderr output: transmission_interface
`

Failed   <<< hardware_interface [1min 0s, exited with code 1]

Summary: 0 packages finished [1min 0s]
  1 package failed: hardware_interface
  1 package had stderr output: hardware_interface
```

Karsten1987, 2021-04-27:
As described in the error message, you're missing the package [ros2_control_test_assets](https://github.com/ros-controls/ros2_control/tree/master/ros2_control_test_assets). That package however is in the same repo.

how are you building your workspace? Make sure to build all packages up to `transmission_interface` to correctly compile all dependencies as well.

newcanopies, 2021-04-28:
Thank you, is there a way to build them **all at the same time**?

I am building them individually, within each package root, and I tried various build sequences ....
In the case of  [ros2_control_test_assets](https://github.com/ros-controls/ros2_control/tree/master/ros2_control_test_assets) it is successfull:

~/ws_ROS2/ros2_control/ros2_control_test_assets$ colcon build
```
Starting >>> ros2_control_test_assets
Finished <<< ros2_control_test_assets [0.14s]

Summary: 1 package finished [0.23s]
```

newcanopies, 2021-04-28:
My mistake, not sure why I was building them individually in the root of each package, 
of course it builds successfully from `ro2_control` root. 

Thank you, this solves the issue.

closed 2021-04-28 by newcanopies