ros-controls/ros2_control #406: Fix INSTANTIATE_TEST_CASE_P for Rolling in Foxy

bmagyar, 2021-04-30:
A fix was proposed to fix INSTANTIATE_TEST_CASE_P for Rolling that also works in Foxy, except for OSX.

https://github.com/ros-controls/ros2_control/pull/390

The issue in this case comes from clang which we can mitigate with a macro check.

vatanaksoytezer, 2021-04-30:
@bmagyar thanks for looking at this! I think mitigating with a macro check makes sense. I will send a PR with the macro checks soon.

vatanaksoytezer, 2021-04-30:
Does something like this looks fine?

```
#ifdef __APPLE__
INSTANTIATE_TEST_CASE_P(
  IdentityMap,
  BlackBoxTest,
  ::testing::Values(
    SimpleTransmission(10.0),
    SimpleTransmission(-10.0),
    SimpleTransmission(10.0, 1.0),
    SimpleTransmission(10.0, -1.0),
    SimpleTransmission(-10.0, 1.0),
    // remove once INSTANTIATE_TEST_SUITE_P is available to use in gtest
    // https://github.com/google/googletest/issues/1419
    // cppcheck-suppress syntaxError
    SimpleTransmission(-10.0, -1.0)), );
  #else
    INSTANTIATE_TEST_CASE_P(
      IdentityMap,
      BlackBoxTest,
      ::testing::Values(
        SimpleTransmission(10.0),
        SimpleTransmission(-10.0),
        SimpleTransmission(10.0, 1.0),
        SimpleTransmission(10.0, -1.0),
        SimpleTransmission(-10.0, 1.0),
        // remove once INSTANTIATE_TEST_SUITE_P is available to use in gtest
        // https://github.com/google/googletest/issues/1419
        // cppcheck-suppress syntaxError
        SimpleTransmission(-10.0, -1.0)));
  #endif
```

bmagyar, 2021-05-14:
How about this?

```
INSTANTIATE_TEST_CASE_P(
  IdentityMap,
  BlackBoxTest,
  ::testing::Values(
    SimpleTransmission(10.0),
    SimpleTransmission(-10.0),
    SimpleTransmission(10.0, 1.0),
    SimpleTransmission(10.0, -1.0),
    SimpleTransmission(-10.0, 1.0),
    // remove once INSTANTIATE_TEST_SUITE_P is available to use in gtest
    // https://github.com/google/googletest/issues/1419
    // cppcheck-suppress syntaxError
#ifdef __APPLE__
    SimpleTransmission(-10.0, -1.0)), );
  #else
    SimpleTransmission(-10.0, -1.0)));
#endif
```

seems easier to maintain

vatanaksoytezer, 2021-05-14:
> How about this?

This gives syntax errors (see https://github.com/ros-controls/ros2_control/pull/416/checks?check_run_id=2584928881), I guess this is the case since this actually a one liner?

matthew-reynolds, 2021-05-14:
The issue is because preprocessor directives within macro arguments are UB. I imagine gcc and clang handle it fine, but cpplint seems to be more strict about not allowing undefined behaviour. Which is good, we obviously shouldn't rely on it.

Something like this should be safe and pretty maintainable:

```
// remove once INSTANTIATE_TEST_SUITE_P is available to use in gtest
// https://github.com/google/googletest/issues/1419
// cppcheck-suppress syntaxError
#ifdef __APPLE__
#define INSTANTIATE_TEST_CASE_P_COMPAT ,
#else
#define INSTANTIATE_TEST_CASE_P_COMPAT
#endif

INSTANTIATE_TEST_CASE_P(
  IdentityMap,
  BlackBoxTest,
  ::testing::Values(
    SimpleTransmission(10.0),
    SimpleTransmission(-10.0),
    SimpleTransmission(10.0, 1.0),
    SimpleTransmission(10.0, -1.0),
    SimpleTransmission(-10.0, 1.0),
    SimpleTransmission(-10.0, -1.0))
    INSTANTIATE_TEST_CASE_P_COMPAT);
```
 (Though not sure I'm happy with the name `_COMPAT`. Maybe `_LAST`? `_END`? I dunno...)

vatanaksoytezer, 2021-05-17:
Thanks for the input @matthew-reynolds ! I think the name `_COMPAT` looks reasonable as well as the solution. I updated the PR with this.

closed 2021-05-18 by bmagyar