ros-controls/ros2_control #421: Release for Galactic and Rolling

henningkayser, 2021-05-25:
What's the status on releasing `ros2_control` for Galactic or Rolling?
We are planning to [release MoveIt](https://github.com/ros-planning/moveit2/issues/476) for both distros end of June and it would be great if this dependency would be available by then.

bmagyar, 2021-06-06:
Hi @henningkayser , we talked with @vatanaksoytezer at the last WG meeting and agreed that we can go ahead with the releases. 

Unfortunately  this is currently blocked on the release of `control_msgs` in Rolling and Galactic with failures on the build farm, here's an example: https://build.ros2.org/job/Gbin_ufv8_uFv8__control_msgs__ubuntu_focal_arm64__binary/19/

I've already asked OR if they know if this is something on the buildfarm or something we can fix but I currently have no bandwidth to deep-dive into an investigation on this.

bmagyar, 2021-06-08:
https://github.com/ros/rosdistro/pull/29786

vatanaksoytezer, 2021-06-08:
Thanks a lot for the update @bmagyar!

Timple, 2021-06-15:
So the `control_msgs` compile fine now on the buildfarm. Could we get a galactic release of `ros2_control`?

vatanaksoytezer, 2021-06-30:
Thanks a ton for rolling @bmagyar: https://github.com/ros/rosdistro/pull/30070!

cco-seasony, 2021-07-02:
Hey! Any updates on the release for Galactic?

bmagyar, 2021-07-04:
https://github.com/ros/rosdistro/pull/30084 ros2_controllers has also been bloomed a few days ago but is being held for a rolling sync

bmagyar, 2021-07-04:
The Galactic release is waiting for one more PR, I need to add parsing for transmissions which will result in some breaking changes - don't want to break things in the next Galactic release... A few more days and we'll have that

tylerjw, 2021-08-11:
Any chance that PR was merged and we could get a Rolling and Galactic release?  I'm just bumping this because it has been a month and I don't see a Galactic release yet.

hakuturu583, 2021-08-28:
Hi, I try to install ros2_control packages in ROS2 galactic.
Do you know current development/release status?

bmagyar, 2021-09-06:
We'd like to get a couple more small ABI breaking changes merged before we release to Galactic. The last Foxy release was made a week ago and we branched for Foxy. Now `master` is for Galactic. 

If you want to help out you can here: https://github.com/ros-controls/ros2_control/issues/314

joe28965, 2022-04-12:
Is it me or has ros2_control since been released on both Galactic and Rolling and should this issue be closed?

bmagyar, 2022-04-12:
Oh yes of course, long since it's been released to both. Closing

closed 2022-04-12 by bmagyar