ros-controls/ros2_control #422: StateInterface with values ​​other than double

anscipione, 2021-05-26:
If I need to have a sensor value other than double (for example an integer array or a string) how should I proceed?

The current implementation seems to allow only double values.

destogl, 2021-06-06:
Hi @anscipione,

For integers, you could simply use doubles for now. (And cast them if needed).

Arrays and strings are currently not supported. We have already ides to support arrays, but strings will take time, because we are still gathering ideas how to handle those in the real-time friendly manner.

anscipione, 2021-06-06:
as for the strings, I needed to pass the frame_id from the definition of the robot. the arrays, on the other hand, I need to pass the readings of some sonar arranged in a circle. for the frame_id I thought I'd pass it as an associated controller parameter. for arrays I'm going to pass them as single _elements_ at the moment.

 thank you

closed 2021-06-06 by anscipione

destogl, 2021-06-07:
> as for the strings, I needed to pass the frame_id from the definition of the robot. the arrays, on the other hand, I need to pass the readings of some sonar arranged in a circle. for the frame_id I [...]

Will it work for you having the string only in controller, for now?
You can pass them there simply using parameter.

anscipione, 2021-06-07:
for now it can be done.  

later they would be useful for me to pass: 
- diagnostic strings, 
- the raw data of the gps (nmea)

Nate711, 2025-02-22:
Does StateInterface still only support doubles? I'd like int64 for reporting timestamps

saikishor, 2025-02-22:
> Does StateInterface still only support doubles? I'd like int64 for reporting timestamps

Hello!

As of now, it is not supported. Soon, we will be adding support for basic types.