ros-controls/ros2_control #437: Can we add string value type?

hakuturu583, 2021-06-07:
Hi, I am developing some hardware interface for ROS2.
I found that current version of ReadOnlyHandle class only supports double type value.

https://github.com/ros-controls/ros2_control/blob/98ab186865d462825e78bfa656c97833b484827d/hardware_interface/include/hardware_interface/handle.hpp#L76

I think if we can add string type here, we can use complex combination of variables easilly by using protobuf.
How do you think about it?
regards,

destogl, 2021-06-08:
No, please let's try to avoid serialization/deserialization inside the library. 

Also regarding string.... we will probably support them at some point, but rather in the form of "limited length char array".  The issue with using string is memory allocation we have to avoid as much as possible in the real-time loop.

hakuturu583, 2021-06-15:
I am now developing realsense t265 tracking camera hardware_interface, and I am now publishing rs2_pose by using ControllerInterface for publising odometry.

https://github.com/OUXT-Polaris/realsense_hardware_interface

I think if we can pass big data such as image/pointcloud by using HardwareStateInterface, we can use ros2_control much more easilly.

At first, I tried to call careate_publisher method inside hardware_interface, but I found it is impossible.
So, I think this method.(using protobuf convert image to string and sending converted string to the ControllerInterface)

hakuturu583, 2021-06-15:
I also agree with constant length double array.
What I want is communicating big data between HardwareInterface and Controller

destogl, 2021-07-20:
@hakuturu583 please look at our meeting notes. We discussed this use-case a few weeks ago, but there is currently no one can start working on this any time soon.

> What I want is communicating big data between HardwareInterface and Controller

what data would you like to pass, and why? Are those data related to real-time control?

closed 2022-06-28 by bmagyar