ros-controls/ros2_control #440: Offer Support for a Simple Transmission with One Actuator and Multiple Joints

niiquaye, 2021-06-14:
The current simple transmission provided by the ros-controls (ros foxy) repo does not support having more than one joint and actuator at a time. I was wondering if a feature like this could be added to make it easier for one to lets say control a 6 wheel robot with two transmissions where each transmission couples 3 joints to one actuator.  Does anyone else have any good ideas on where to start. This new feature can then greatly assist setting up a diff drive controller for a robot with said restrictions above. I am new to open source development - any and all feedback is greatly appreciated.

niiquaye, 2021-06-14:
[image]

destogl, 2021-06-14:
Hi @niiquaye, the question is, do you really need transmission to multiple actuators here or you can model it as two wheels diff-drive robot and then just send data to multiple actuators in your hardware component. Maybe you could provide more details about the setup/hardware interfaces you are using to help you more concretely.

niiquaye, 2021-06-14:
Hi @destogl thank you for the quick response. The setup I am going with is controlling a 6 wheel robot with two [Odrives](https://odriverobotics.com/). The catch is there will be a reducing transmission paired to each of the Odrives. Then each Odrive will in turn drive 3 wheels. I wanted to try and get an accurate representation of the urdf model of this 6 wheel robot vs a 2 wheel diff drive. It may very be possible to view the problem as a two wheel robot (we use CAN messages that will get sent in our overloaded write function from ros2-control system interface). Would setting up the problem as a two wheel diff drive robot vs 6 wheel diff drive robot work nicely in a real world setting using a differential controller (As the differential controller uses wheel separation and number of wheels in its calculations of left and right velocities)? Or would you suggest making a transmission that can support this new feature?