ros-controls/ros2_control #445: hardware_interface's destructor is not called when I exit with Ctrl + C during MoveIt's trajectory playing.

ShotaAk, 2021-06-24:
When `Ctrl+C` is pressed while playing the trajectory output by MoveIt, the destructor of my robot's hardware_interface is not called.

This can be reproduced by pressing `Ctrl+C` after clicking `Execute` button of the RViz panel.

[image]

The destructor turns off the motor torque of my robot, so if this destructor is not called, the motor will remain running.

https://github.com/rt-net/crane_plus/blob/5875fb5002d477d1c969c7487c3a41a5028288ec/crane_plus_control/src/crane_plus_hardware.cpp#L28-L32

I edited the destructor of `ControllerManager` and released `resource_manager` as following.
As a result, the destructor of hardware_interface is now called even when Ctrl + C is pressed during trajectory playback.

https://github.com/ros-controls/ros2_control/blob/630b61099f8bbd473de6727c9aa351833ae5de30/controller_manager/include/controller_manager/controller_manager.hpp#L68-L70

```diff
  CONTROLLER_MANAGER_PUBLIC
-  virtual
-  ~ControllerManager() = default;
+  ~ControllerManager(){
+    RCLCPP_INFO(rclcpp::get_logger("ControllerManager"), "Destructor has called.");
+    resource_manager_.reset();
+ }
```
Please let me know if there is an easier way to solve this problem.

Environments:
- ROS : Foxy
- OS : Ubuntu 20.04
- ros2_control : 630b61099f8bbd473de6727c9aa351833ae5de30
- ros2_controllers : 54dd7837dc0cff1445eb06f36a152d3d3bd41bb1
- moveit2 : eb2c8b97862e9acb760dd0c7882d06d04e4ecfae

bmagyar, 2021-06-25:
Very nice analysis, I think I can tell you what to do based on this!

In order for derived-classes' destructors to be called properly when taken down through a base class interface, the destructor needs to be declared `virtual`. 
Here: 
https://github.com/rt-net/crane_plus/blob/5875fb5002d477d1c969c7487c3a41a5028288ec/crane_plus_control/include/crane_plus_control/crane_plus_hardware.hpp#L45

Try that and let us know if that fixed your problem

ShotaAk, 2021-06-28:
Thank you for the advice.
Unfortunately, the result did not change.

I edited the destructor as follows.
```diff
  CRANE_PLUS_CONTROL_PUBLIC
- ~CranePlusHardware();
+ virtual ~CranePlusHardware();
```

Here is the log when I quit with Ctrl+C.
```txt
[move_group-1] [INFO] [1624872271.796853940] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: crane_plus_arm_controller started execution
[move_group-1] [INFO] [1624872271.796885903] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[rviz2-2] [INFO] [1624872272.035784818] [move_group_interface]: Plan and Execute request accepted
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[move_group-1] [INFO] [1624872273.110592912] [rclcpp]: signal_handler(signal_value=2)
[move_group-1] [WARN] [1624872273.111472170] [moveit_ros.plan_execution]: Stopping execution due to unknown reason.Possibly the node is about to shut down.
[move_group-1] [INFO] [1624872273.111506048] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Cancelling execution for crane_plus_arm_controller
[static_transform_publisher-3] [INFO] [1624872273.110593246] [rclcpp]: signal_handler(signal_value=2)
[robot_state_publisher-4] [INFO] [1624872273.110602128] [rclcpp]: signal_handler(signal_value=2)
[ros2_control_node-5] [INFO] [1624872273.110668399] [rclcpp]: signal_handler(signal_value=2)
[rviz2-2] [INFO] [1624872273.115507081] [rclcpp]: signal_handler(signal_value=2)
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 117961]
[INFO] [static_transform_publisher-3]: process has finished cleanly [pid 117959]
[ros2_control_node-5] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-5]   what():  Asked to publish result for goal that does not exist
[rviz2-2] [WARN] [1624872273.131613882] [interactive_marker_display_94242158054096]: Server not available while running, resetting
[rviz2-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rviz2-2]   what():  Failed to create interrupt guard condition in Executor constructor: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/guard_condition.c:67
[ERROR] [ros2_control_node-5]: process has died [pid 117963, exit code -6, cmd '/home/shotaak/moveit_ws/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_65l2xwp1 --params-file /home/shotaak/ros2_ws/install/crane_plus_control/share/crane_plus_control/config/crane_plus_controllers.yaml'].
[ERROR] [rviz2-2]: process has died [pid 117957, exit code -6, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/shotaak/ros2_ws/install/crane_plus_moveit_config/share/crane_plus_moveit_config/launch/run_move_group.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_npiblzk7 --params-file /tmp/launch_params_ubzkulv7 --params-file /tmp/launch_params_3g7h6r1e --params-file /tmp/launch_params_wrymhgc2'].
[move_group-1] [WARN] [1624872275.929233179] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: waitForExecution timed out
[ERROR] [move_group-1]: process[move_group-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [move_group-1]: sending signal 'SIGTERM' to process[move_group-1]
[ERROR] [move_group-1]: process has died [pid 117955, exit code -15, cmd '/home/shotaak/moveit_ws/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_rpixbph8 --params-file /tmp/launch_params_mryvme9v --params-file /tmp/launch_params_ab9gyufu --params-file /tmp/launch_params_rj_dhj0l --params-file /tmp/launch_params_mo2x884m --params-file /tmp/launch_params_okgnudss --params-file /tmp/launch_params_s7iooz9g'].
[move_group-1] 

```

ShotaAk, 2021-07-02:
Update:

This problem occurs when I press Ctrl + C while the trajectory is "playing".

When MoveIt is doing nothing, the destructor is called successfully. (e.g., before and after trajectory playback)

Karsten1987, 2021-07-02:
I somewhat confident that `terminate called after throwing an instance of 'std::runtime_error'` is your real problem here. Somewhere along your stack there's an unhandled exception leading to terminate. Terminate provokes no stack windup and hence your destructors are not called.
Would be interesting to know why the node throws an exception - most likely it's the JTC which raises the error.

ShotaAk, 2021-07-05:
Is the cancellation process of the FollowTrajectoryAction not working properly ? :thinking: 
(JTC stands for JointTrajectorryController, right?)

bmagyar, 2021-07-05:
Following a few discussions spawning from this issue I've created 3 follow-up issues, each of which should help mitigate issues like this in the future.

closed 2022-07-16 by bmagyar