ros-controls/ros2_control #446: Build issues

MarqRazz, 2021-06-28:
I am trying to debug an issue with MoveIt2 but when I try to build with `--cmake-args -DCMAKE_BUILD_TYPE=Release`/`Debug` or use  `--mixin rel-with-deb-info` with ros2_control in my workspace I get the following errors. I am running Foxy with everything up to date on `master`

```
--- stderr: joint_state_broadcaster                                                    
CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:39 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)

CMake Error at /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/ament_cmake_export_dependencies-extras.cmake:43 (get_target_property):
  get_target_property called with incorrect number of arguments
Call Stack (most recent call first):
  /home/marq/ros_control2_ws/install/controller_manager/share/controller_manager/cmake/controller_managerConfig.cmake:41 (include)
  CMakeLists.txt:58 (find_package)
---
```

bmagyar, 2021-07-05:
If you are debugging moveit2, why have ros2_control in your workspace? Just grab 'em from the apt repos.

MarqRazz, 2021-07-07:
I was able to work around the issue, but mainly posted it here to point out that you cant build ros2_control with `Debug`. Feel free to close the issue.

bmagyar, 2021-07-08:
I was able to build it, had `ros2_control`, `ros2_controllers`, `ros2_control_demos` and `gazebo_ros2_control` in the workspace :+1:

MarqRazz, 2021-07-08:
I pulled everything up to date and can now build the same packages with `Debug` and `--mixin rel-with-deb-info`. Great work guys and thanks for your help!

closed 2021-07-08 by MarqRazz