ros-controls/ros2_control #448: Velocity Controller does not operate in Gazebo when position command interface is also specified.

GusBus4, 2021-06-30:
Adding in a position command interface in the ros2_control tag inside the URDF stops the velocity controller from operating.
Position and Joint Trajectory Controllers still operate fine.

More Details: 
Ubuntu 20.04
ROS2 Foxy
Build ros2-control from source Main Branch (pulled 6/30/21)

Launch File Steps:
- Opens Gazebo
- Creates Robot State Publisher Node
- - 1 Continuous Joints and 2 Revolute Joints
- - Each with Position and Velocity command interfaces
- Spawns Robot
- Loads Position State Broadcaster (Active)
- Loads Position Trajectory Controller (Inactive)
- Loads Position Group Controller (Inactive)
- Loads Velocity Controller (Inactive)

Everything loads correctly, list_controllers shows all controllers in the correct state. Hardware interfaces shows all has unclaimed. 

If I then start the velocity controller (Hardware interfaces show has claimed) and publish commands the robot does not move. If I switch to position of trajectory behavior is as expected. Even if I completely unload the positions and trajectory controllers the robot does not adhere to velocity control.  Removing the position and trajectory controllers from the YAML config and never loading them in the first place also does not improve the situation.

If I remove the position command interfaces tag from the URDF, then the velocity controllers operates as expected.

GusBus4, 2021-07-07:
Further investigation shows that that the issue lives with gazebo_ros2_control plugin and not ros2_control
Closing this issue and reopening in gazebo_ros2_control repo.

Essentially the problem boils down to gazebo_ros2_control plugin setting positions, velocities and efforts based on which command interfaces are defined in the ros2_control tag and not if that specific hardware interface has been allocated or not.

closed 2021-07-07 by GusBus4

gavanderhoorn, 2021-07-07:
What's the issue on the `gazebo_ros2_control` tracker?

GusBus4, 2021-07-07:
@gavanderhoorn Sorry I was still creating it. Here is the link to the issue.

https://github.com/ros-simulation/gazebo_ros2_control/issues/85