ros-controls/ros2_control #450: Add terminate handler to ros2_control_node

bmagyar, 2021-07-05:
A terminate handler should be added to `ros2_control_node` as a last resort to ensure destructors of system components are called and (when done) everything in `ResourceManager` can be put through their lifecycles properly.

Implementation idea:
* ResourceManager and ResourceStorage could have some way of resetting
* which should be triggered from `terminate_handler` defined in `ros2_control_node`.

This should help with issues similar to #445 in the future.

bmagyar, 2021-07-07:
@Karsten1987 @destogl @v-lopez @mahaarbo

saikishor, 2025-07-08:
@christophfroehlich this is solved by the callback we added so all the components and controllers shutdown before exiting right?