ros-controls/ros2_control #451: Exceptions thrown in controllers should never crash the entire framework

bmagyar, 2021-07-05:
Restrict the scope of exceptions thrown in controllers to those controllers.

General idea: When an exception is thrown in a controller, that controller should go into some failed state. This should impose no additional changes on existing controller code.

Implementation ideas:
* Add some hooks into controller_interface to do the catching?
* Add a general try-catch around `update()` in `controller_manager` and manage controller state? (Also lays itself well for later to put an entire chain of controllers into error state if one of them fails)

#445

destogl, 2021-07-05:
I was just looking at it because my controller is crashing everything :D

We can cover all in controller_interface except `init` call. This is really propagated through inheritance. For other methods, we have on_* controller methods.

bmagyar, 2021-07-07:
@Karsten1987 @v-lopez @mahaarbo

v-lopez, 2021-07-08:
I'd go with the general try-catch.

If someone is careful enough to implement a hook that does the exception processing, he should probably catch them at the source.

RandomSpaceship, 2024-12-19:
@christophfroehlich and @bmagyar should this issue be closed now given #1507 is merged? Looks like GitHub didn't automatically link the issue to the PR.

saikishor, 2024-12-19:
@RandomSpaceship Thanks for bring it up. You are right. This can be closed.

closed 2024-12-19 by saikishor