ros-controls/ros2_control #452: Time to get rid of exceptions

bmagyar, 2021-07-05:
We have error codes returned as well as exceptions.

While it's quick and easy to flag up issues this way while things are wildly developing, I think we have reached a more mature state where it's time to start pruning them. Of course there are places where they may still be justified.

Opinions?
@Karsten1987 @destogl @v-lopez @jordan-palacios

v-lopez, 2021-07-05:
Why are return codes preferred over exceptions? As a general rule I prefer exceptions for most code, as they are harder to ignore, and more verbose.

Also is it a blanket ban on exceptions? 
Also how are we going to return errors on functions such as `load_controller` (https://github.com/ros-controls/ros2_control/blob/master/controller_manager/include/controller_manager/controller_manager.hpp#L74)

I apologize if this was discussed in one of the many meetings I have missed.

bmagyar, 2021-07-07:
The title may be a little extreme :) 

> Also is it a blanket ban on exceptions?
> > Of course there are places where they may still be justified.

Indeed they are harder to ignore but currently they simply create a mess and we do error codes and exceptions which feels like twice the work. This is no global ban or anything like that, simply a discussion opener. If you have good general practices, feel free to add them here.

Karsten1987, 2021-07-08:
exceptions are generally not RT safe and have further implications with them, so I am personally in favor of using return codes. What we do quite a fair deal in the ros2 core packages is to flag certain functions with attributes to give you compiler warnings/errors when a return code is ignored: https://github.com/ros2/rcutils/blob/master/include/rcutils/macros.h#L25-L31