ros-controls/ros2_control #464: controller manager doesn't return the controller's joints when it's inactive

JafarAbdi, 2021-07-15:
I'm trying to port `moveit_ros_control_interface` to ROS2 and it has an option to let moveit's trajectory execution manager automatically switch controllers depend on the joints in the received trajectory, but the problem we're having is ros2 controller manager doesn't return the joints name for the inactive controller, 

I'm trying to fix it and looking at the code it seems like it's due to this [line](https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/controller_manager.cpp#L583), should I open a PR that fixes it upstream .?

destogl, 2021-07-20:
I think it is not simply that line. Actually, at current state, the CM has no idea about joints and interfaces of inactive controllers. This all happens when a controller tries to be activated. Having this option would need some more work to internally store the interfaces a controller need. I would be also feasible to get those values when the service is called, but this could prolong response time on the service...

schornakj, 2021-08-24:
I'm also interested in allowing the `moveit_ros_control_interface` to get info about which interfaces are associated with currently-inactive controllers. @destogl how fundamental of a change to the implementation of the controller manager would this be? Has there been more detailed discussion since @JafarAbdi's initial post about where modifications would need to be made?

closed 2021-10-15 by bmagyar