ros-controls/ros2_control #472: Proper way to shutdown the hardware_interface

MarqRazz, 2021-07-20:
Hello,

I'm implementing a driver for a new robot arm and have questions on how to properly shutdown the hardware interface node. When the user kills the node I want to return the control mode of the robot controller to a specified state and close communication with it. While I was implementing this I noticed if you shut down with `ctrl-c` the `stop()` method is not called. Should it be?

To fix this I added
`rclcpp::on_shutdown(std::bind(&MultiInterfaceHardware::stop, this));`
to my hardware interface and now it properly sets the controller to a safe mode and then closes communication. Is this the proper way to implement this or should we consider making this the default behavior and have the framework call the user defined `stop()` method?

jaron-l, 2021-07-29:
Can you ask this question on answers.ros.org? That's the best way to get answers as the [documentation](https://ros-controls.github.io/control.ros.org/#development-organisation-and-communication) suggests. It's the way most questions for ROS are funneled for documentation and so there's a larger community to help out. Be sure to use the `ros2_control` tag so it's easy for maintainers to see.

roncapat, 2021-08-25:
Hello @MarqRazz , have you found a solution on how to properly shut down the Controller Manager? 

I have noticed too that stop() method is never used, in fact ros2_control/controller_manager/src/ros2_control_node.cpp (indirecly thru ros2_control/controller_manager/src/controller_manager.cpp) does never call stop_components().

Let me know if you found a solution or opened a ticket on the answers.ros.org website. Thank you :)

lukicdarkoo, 2022-03-28:
It looks like a bug and it is still present. The `on_deactivate` callback is not called on `CTRL-C`. @MarqRazz's workaround works though.

bmagyar, 2022-03-29:
I'll try to set up a test for this and add a fix.

swiz23, 2023-01-29:
Still seeing this issue in January of 2023.

closed 2025-01-29 by bmagyar