ros-controls/ros2_control #486: Modular xacros can spawn multiple copies of the same hw_interface

AndyZe, 2021-08-05:
As described here on [ROS-answers](https://answers.ros.org/question/383824/ros2_control-combined-hw_interface-for-arm-and-gripper/?answer=383854#post-id-383854):

It's common to have a modular urdf. There will be a high-level "system" urdf that pulls in an arm xacro as well as a gripper xacro.

I have learned, there can only be one ros2_control xacro, otherwise it spawns 2 identical hw_interfaces. So there cannot be individual ros2_control xacros for the gripper and the arm.

This isn't a huge deal but it's not an ideal situation for, say, switching between many end-effectors.

destogl, 2021-08-09:
Hi,

this is not a bug is a feature :D

Generally, if you define multiple tags with the same hardware interface, this will be started multiple times.

Let me explain the idea behind this and what was a mistake you did.

You should have one `ros2_control` tag per hardware interface. In your case, you have a robot and gripper which have the same hardware interface so they should be described in one tag. This use-case is described/shown in Example 3 in [roadmap](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#3-industrial-robots-with-integrated-sensor) and [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos/pull/112) (open PR). We are using the sensors in the example, but it is the same for an integrated gripper (as in your case).

If you need to use separate hardware interfaces for your components, i.e., arm and gripper, then you will use multiple `ros2_control` tags as shown in Example 4 in [roadmap[(https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#4-industrial-robots-with-externally-connected-sensor) and [ros2_control_demos](https://github.com/ros-controls/ros2_control_demos/pull/120) (work in progress).

This means that in your case doesn't make much sense to split `ros2_control` tag across multiple files because they use the same hardware_interface implementation. In this case, one should use parameters.

> This isn't a huge deal but it's not an ideal situation for, say, switching between many end-effectors. Each end-effector might have its own package.

this should work with multiple `ros2_control` tags because it means that each end-effector has its own hardware_interface plugin and they are separated from the arm's communication.

Do you find this approach we chose OK, or do you have another proposal?

AndyZe, 2021-08-09:
I think this is such a rare edge case it's not worth worrying about. I'm working with the Kinova arm and gripper, which are unusual in that the arm and gripper must have the same hw interface.

For 99% of robots out there, it won't matter. Separate hw interfaces are fine.

closed 2021-08-09 by AndyZe

byeongkyu, 2021-10-05:
@destogl May I ask a question? It seems that each hardware_interface (external_sensor and robot) have same control rate. But many robots have different control rate per group of actuator or group of sensors. In this case, I think the robot should has multiple controller_manager. It is right??

destogl, 2021-10-05:
@byeongkyu simply no. You can now specify specific control rate for a controller, but the hardware read and write will be calls with the main (fastest) update rate. If your communication interface does not support that, then you should use a separate thread for hardware communication inside the hardware interface.

byeongkyu, 2021-10-07:
@destogl I got it. Thanks.