ros-controls/ros2_control #487: Starting gripper controller and arm controller --> joint confusion

AndyZe, 2021-08-05:
We are starting a gripper controller and an arm trajectory controller simultaneously. `controllers.yaml` given below. When the controllers try to start, the joints are confused like so. To me, this seems like a parsing issue in the resource_manager?

```
[ros2_control_node-1] [ERROR] [1628170893.039272048] [resource_manager]: Component 'KortexMultiInterfaceHardware' did not accept new command resource combination: 
[ros2_control_node-1]  Start interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1]   joint_1/position
[ros2_control_node-1]   joint_2/position
[ros2_control_node-1]   joint_3/position
[ros2_control_node-1]   joint_4/position
[ros2_control_node-1]   joint_5/position
[ros2_control_node-1]   joint_6/position
[ros2_control_node-1]   joint_7/position
[ros2_control_node-1] ]
[ros2_control_node-1] Stop interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1] ]
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1628170893.039285684] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
[ros2_control_node-1] [ERROR] [1628170893.039388955] [resource_manager]: Component 'KortexMultiInterfaceHardware' did not accept new command resource combination: 
[ros2_control_node-1]  Start interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1]   finger_joint/position
[ros2_control_node-1] ]
[ros2_control_node-1] Stop interfaces: 
[ros2_control_node-1] [
[ros2_control_node-1] ]
[ros2_control_node-1] 
[ros2_control_node-1] [ERROR] [1628170893.039406176] [controller_manager]: Could not switch controllers since prepare command mode switch was rejected.
```

Controller yaml:
```
controller_manager:
  ros__parameters:
    update_rate: 600  # Hz

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    hand_controller:
      type: forward_command_controller/ForwardCommandController

joint_trajectory_controller:
  ros__parameters:
    joints:
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
      - joint_7
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.0
      goal_time: 0.0

hand_controller:
  ros__parameters:
    joints:
      - finger_joint
    interface_name: position
```

AndyZe, 2021-08-05:
Resolved by deleting this check in the hw_interface:   https://github.com/ros-controls/ros2_control_demos/blob/9a43e5d409a21157d1cff819c2988546a7a503b0/ros2_control_demo_hardware/src/rrbot_system_multi_interface.cpp#L185

I guess control_lvl_ was not initialized right.

closed 2021-08-05 by AndyZe

bmagyar, 2021-08-05:
Could you elaborate on why you think this is a parsing issue?

Seeing `KortexMultiInterfaceHardware` rejecting a switch I think it's something that has to do with that component. The way we implemented mode switching is that essentially components need to sign off on it if they are happy:
https://github.com/ros-controls/ros2_control/pull/348/files#diff-31aaa609a55e1333d6e694aea6963bdaceed8615945c65fb57daa91a7c9fd58dR382