ros-controls/ros2_control #489: Simplified `ınit`-method handling in controller_interface for users implementing custom controllers

destogl, 2021-08-07:
Methods controlling the lifecycle of controllers have all prefix `on_` in their implementation (e.g., `on_configure`, `on_activate`).
The only exceptions are `init` and `update` methods.
For the `init` method, this creates a cumbersome implementation where [controller implementation always has to call the parent `init` method](https://github.com/ros-controls/ros2_controllers/blob/a86a8c24b9543035f9db3458d4de51a37c6ed5d7/forward_command_controller/src/forward_command_controller.cpp#L40-L43) (from `ControllerInterface`) and pass its parameters.
The `update` method does not need any changes. 

After discussion with @bmagyar, I am proposing the following changes as of galactic to resolve this/make the code cleaner.

* [ ] Add `on_init()` virtual method in the `controller_interface.hpp` file.
* [ ] Call this method from the `init(const std::string & controller_name)` in `controller_interface.cpp` - (called from ControllerManager)
* [ ] Keept the method overridable (virtual) to enable users to provide custom NodeOptions
* [ ] Remove `init(const std::string & controller_name, rclcpp::NodeOptions & node_options)`
* [ ] Extend test-controllers from `controller_manager` package and tests of controller manager (if needed)
* [ ] Update implementations for all controllers from ros2_controller package: rename `init(const std::string & controller_name)` to `on_init()` and remove the lines similar to [this](https://github.com/ros-controls/ros2_controllers/blob/a86a8c24b9543035f9db3458d4de51a37c6ed5d7/forward_command_controller/src/forward_command_controller.cpp#L40-L43).
* [ ] Updte tests for controllers if needed

Note: It is possible that I missed a few additional minor changes in this list.

Unfortunately, currently,  I don't have the capacity to tackle this, although this is extremely important because it makes code cleaner and simple to use. @jaron-l or @bailaC, do you have some time to tackle this issue? It is a bit more complex to be "good-first-issue", but it is fairly simple for someone with a bit of experience with the framework.

bmagyar, 2021-08-07:
FYI @AugusteBourgois

bailaC, 2021-08-09:
@destogl , @bmagyar I will take care of this.

Karsten1987, 2021-08-09:
Please note that I think it makes sense to change the internal `node` of each controller to a lifecycle node and disable the external service calls

destogl, 2021-08-12:
> Please note that I think it makes sense to change the internal `node` of each controller to a lifecycle node and disable the external service calls

They are already using this interface. As soon as we branch out for "foxy" we can make LifecycleNode chnages.

closed 2021-09-07 by bmagyar