ros-controls/ros2_control #493: Clean up controller manager tests

bmagyar, 2021-08-13:
## Background

A set of easy-to-tackle points were raised on #476 . The following list captures them:

1.
```
 controller_manager/test/test_load_controller.cpp

TEST_F(TestLoadController, can_configure_loaded_controller)
@destogl destogl yesterday Member

Isn't this the same as the load_and_configure_one_known_controller test? Can we merge those two tests?
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687736180

2.
```
 controller_manager/test/test_load_controller.cpp
  // load and start controller with name1
  auto controller_if = cm_->load_controller(controller_name1, TEST_CONTROLLER_CLASS_NAME);
  ASSERT_NE(controller_if, nullptr);
  EXPECT_EQ(
    cm_->configure_controller(controller_name1), controller_interface::return_type::OK);

Comment on lines +164 to +168
@destogl destogl yesterday Member

Would it make sense to place this section into a function because it is repeated almost in all tests? Does this make sense?
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687737563

3.
```
 controller_manager/test/test_load_controller.cpp
}

TEST_F(TestLoadController, switch_controller_empty)
TEST_F(TestLoadController, inactive_controller_cannot_be_configured)
@destogl destogl yesterday Member

This test should be moved up before can_not_configure_active_controller, or the can_not_configure_active_controller should be at least after can_stop_active_controller.

In that was the tests would be like: “normal lifecycle”, "errors in lifecycle"
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687745887

4.
```
 controller_manager/test/test_load_controller.cpp
  auto controller_if = cm_->load_controller(controller_name1, TEST_CONTROLLER_CLASS_NAME);
  ASSERT_NE(controller_if, nullptr);
  EXPECT_EQ(
    cm_->configure_controller(controller_name1), controller_interface::return_type::OK);
@destogl destogl yesterday Member

Also, after this line we should try to configure it again. (and as well after stopping as done later.)
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687746756

5.
```
 controller_manager/test/test_load_controller.cpp
  ASSERT_EQ(
    lifecycle_msgs::msg::State::PRIMARY_STATE_INACTIVE,
    controller_if->get_current_state().id());
  EXPECT_EQ(1u, cleanup_calls);
@destogl destogl yesterday Member

Why is here cleanup called?
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687748385

6.
```
      controller_interface::return_type::ERROR, STRICT, NONEXISTENT_CONTROLLER,
      VALID_CONTROLLER, "Switch with valid stopped controller specified"),
    std::make_tuple(
      controller_interface::return_type::OK, BEST_EFFORT, NONEXISTENT_CONTROLLER,
      VALID_CONTROLLER, "Switch with valid stopped controller specified"),
Comment on lines +420 to +443
@destogl destogl yesterday Member

Error messages do not seem to be right, at least they are not clear. What about to make them longer and more descriptive?
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687753065

7.
```
 controller_manager/test/test_load_controller.cpp
));
#endif

TEST_F(TestLoadController, starting_and_stopping_a_controller)
@destogl destogl yesterday Member

Isn't this test done already above? I think it was "test_configure_inactive_controller".
```
https://github.com/ros-controls/ros2_control/pull/476#discussion_r687756176

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_control/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_control`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_control` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 📝 **Update** the file(s) `$FILENAME`by applying the change shown in the diff below.

```diff
$DIFF
```

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/foxy/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

NourS-d, 2021-09-04:
I think I can take this on

bmagyar, 2021-09-05:
Thanks @NourS-d ! If you have any questions feel free to ask here or tag along to one of the Working Group meetings (next one is next Wednesday) to discuss

christophfroehlich, 2025-07-26:
Seems to be fixed with #532