ros-controls/ros2_control #521: Add `BEST EFFORT` controller switch tests for Controller Manager

destogl, 2021-09-07:
We could probably use parameterized tests for this to save some code.

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/461#issuecomment-914036732_

Xi-HHHM, 2021-11-11:
Please assign to me.

christophfroehlich, 2025-04-03:
> @Xi-HHHM and I talked yesterday about further steps and that would be:
> 
> - [ ] split current test into "lifecycle test" and "activating multiple controllers when one is unknown"
> - [ ] add test for "activating multiple controllers when one has a resource collision"

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/issues/582#issuecomment-1029931561_

ashwinsnambiar, 2025-04-06:
@christophfroehlich Can I take up this issue? I am not very familiar with building tests, but I would like to take this opportunity to learn and contribute to the community.

christophfroehlich, 2025-04-06:
Very welcome! Writing tests is mostly understanding the intention of the code itself, and copy-pasting other tests ;) 
Don't hesitate to ask if something is unclear (I have to read up on this topic as well, I haven't looked into detail yet). But maybe @saikishor can also help fast here.

saikishor, 2025-04-06:
Hello!

https://github.com/ros-controls/ros2_control/blob/master/controller_manager%2Ftest%2Ftest_controller_manager_srvs.cpp

In the above file, there are some tests already present. Better take a look at some coverage builds and understand what else needs to be implemented. I already see some tests with BEST_EFFORT.

ashwinsnambiar, 2025-05-24:
Hey @saikishor  @christophfroehlich, sorry for the long delay. I have some questions about the second test for "activating multiple controllers when one has a resource collision". I was thinking of creating a parametric test for both strictness values. I started three controllers, out of which one and two are in resource collision. As expected, in the `STRICT` mode, the test failed immediately. I was expecting the controllers in `BEST_EFFORT` mode to remain in blocking until next update cycle, but instead it seems to fail immediately. Is this the way its supposed to work? Is my understanding of the working of the controllers wrong somewhere? Could someone review the PR?