ros-controls/ros2_control #527: colcon build fails at controller_manager

Betancourt20, 2021-09-08:
Hi,
I'm trying to build Moveit in Ubuntu 20.04 and I already install ROS 2 foxy. Everything was okay but I have an error which I tried to search if some had the same problem as me, but I did not find anything. I leave the error in my console :

Finished <<< controller_manager_msgs [5.83s]
Starting >>> controller_manager
--- stderr: controller_manager
/home/unknown/ws_moveit2/src/ros2_control/controller_manager/src/controller_manager.cpp: In function ‘bool controller_manager::is_controller_inactive(const controller_interface::ControllerInterface&)’:
/home/unknown/ws_moveit2/src/ros2_control/controller_manager/src/controller_manager.cpp:34:21: error: ‘const class controller_interface::ControllerInterface’ has no member named ‘get_state’
34 | return controller.get_state().id() == lifecycle_msgs::msg::State::PRIMARY_STATE_INACTIVE;
| ^~~~~~~~~
make[2]: *** [CMakeFiles/controller_manager.dir/build.make:63: CMakeFiles/controller_manager.dir/src/controller_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:193: CMakeFiles/controller_manager.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

Failed <<< controller_manager [5.93s, exited with code 2]
Aborted <<< moveit_msgs [17.0s]

Summary: 16 packages finished [17.7s]
1 package failed: controller_manager
1 package aborted: moveit_msgs
1 package had stderr output: controller_manager
36 packages not processed

Does anyone has any clue to solve it?
Thanks in advance

bmagyar, 2021-09-08:
Please use the foxy branch. The `master` branch has moved on to development for Galactic.

bmagyar, 2021-09-08:
I updated the readme to avoid future confusion: https://github.com/ros-controls/ros2_control/commit/7a0d5169c6a0f4aa8381b0e57de22ee28e6172bf

bmagyar, 2021-09-08:
Plus there's this: https://github.com/ros-controls/ros2_control/pull/526

closed 2021-09-08 by destogl