ros-controls/ros2_control #531: Reconsider `automatically_declare_parameters_from_overrides` for all controllers

destogl, 2021-09-09:
Long story short: "I have a follow-up issue with this parameter"

_Originally posted by @destogl in https://github.com/ros-controls/ros2_control/pull/504#r705324464_

Long story:

I understand that we need this automatic declaration for JTC per-joint trajectory limits. And adding this parameter caused further fix in form of the `auto_declare` method because otherwise, the controllers are declaring parameters twice.

Now I have an issue when using external libraries which declare their parameters, like [ros/filters](https://github.com/ros/filters/tree/ros2) library, especially filters chain structure. To get this working we would need to add an `auto_declare` throughout rclcpp. this leads me to the conclusion that the approach done in #504 is probably not the right one and we should reconsider it again.

destogl, 2021-09-13:
No, because they are not a node on their own. And it shouldn't be. I think that we don't have this issue if we properly declare parameters in the JTC. This was the source of this issue.

bmagyar, 2021-09-15:
It seems we are alternating between two bad solutions. Is there a third one that's clean??

destogl, 2021-09-16:
I don't think so. I am actually sure that the current approach is not correct for our library. This is OK if you have some simpler node without possiblity to extend it with many libraries. The old approach is right and we should simply declare those parameter explicity. As done [here](https://github.com/ros-controls/ros2_controllers/blob/60db804d694743a86edb2b07352b08e193e23698/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L66-L71) (ros-controls/ros2_controllers#225)

livanov93, 2021-09-20:
I agree the current solution is not the cleanest, however in my understanding the solution that existed before was aiming towards current behavior (avoiding the need to explicitly declare every parameter for the controller's use). However, as @destogl  says, incorporation of other packages that don't use 'auto_declare' could really be a problem here. 
We can then remove `automatically_declare_parameters_from_overrides` flag in the controller interface [here](https://github.com/ros-controls/ros2_control/blob/ac3b10ea20104660f9ebf25b66ec04091773e086/controller_interface/src/controller_interface.cpp#L32) and do all declaration explicitly. 
In the future we have to read rclcpp docs more carefully regarding parameter propagation [here](https://github.com/ros2/rclcpp/blob/2801553d61c5a30a0327d5cbc8d28bcd74e9703d/rclcpp/include/rclcpp/node_options.hpp#L337) and [here](https://github.com/ros2/rclcpp/blob/2801553d61c5a30a0327d5cbc8d28bcd74e9703d/rclcpp/include/rclcpp/node_options.hpp#L360).

However `auto_declare` could be useful  as node's auxiliary method which is more of a topic for [ros2/rclcpp](https://github.com/ros2/rclcpp) repository.

destogl, 2021-09-24:
@livanov93 if you have some capacity, this would be great. I think that only JointTrajectoryController has these issues.

livanov93, 2021-10-04:
@destogl  I will address it soon and let hardcode the parameter declaration for jtc.

closed 2022-07-16 by bmagyar