ros-controls/ros2_control #535: Colored output of ros2_control

fmros, 2021-09-18:
On ros_control for ROS1 i was wishing for some colored output for successful loaded and started controllers on the terminal. 
Should we introduce that on ros2_control, for selected outputs?
E.g. A green output for "Configured and started controller". Does somebody share this favor? 

Let me know, I can create a PR.

bmagyar, 2021-09-19:
I'm all for it! :D
In various pieces of code I used to define some extra colour macros like this:
https://github.com/bmagyar/unknown_object_segmentation/blob/fc7571511e69851797fd6894c35d7310ae62c32a/src/segmenter_node.cpp#L8

fmros, 2021-09-19:
> I'm all for it! :D
> In various pieces of code I used to define some extra colour macros like this:
> https://github.com/bmagyar/unknown_object_segmentation/blob/fc7571511e69851797fd6894c35d7310ae62c32a/src/segmenter_node.cpp#L8

Yeah i used similar techniques in the past. Shall i go for it or wait for additional feedback?

bmagyar, 2021-09-19:
let's hear from @destogl

destogl, 2021-09-24:
YES! Please open PR :cool: