ros-controls/ros2_control #547: Is ros2_control support multiple controller_manager?

byeongkyu, 2021-10-01:
Is ros2_control support_multiple controller_manager?
for example, each controller_manager has different joints and different control rate.

bmagyar, 2021-10-01:
I think this came up earlier and it was possible to namespace the
controller manager however I'd add that that is more for a multi robot
usecase, I really wouldn't recommend that within one robot. If you have a
specific use case where the hardware is multi rate,etc,that's something we
can extend the framework to handle.

byeongkyu, 2021-10-01:
For example, in my robot. It has two communication lines.  One is CAN, other is RS485. Each communication has different control rate and different init method.

bmagyar, 2021-10-05:
Why not implement them as different system components?

byeongkyu, 2021-10-05:
I’ll use them as hardware_interface because I’ll use exist controller (diff_drive_controller, forward_command_controller, etc.).  As I understand, it probably need two or more controller_manager and it need each  URDF <ros2_control> files. Is it right?

destogl, 2021-10-05:
@byeongkyu  You can use multiple hardware components inside one controller manager. We currently don't support reading and writing at different rates, but this shouldn't be the issue since you can use different reading rates inside a hardware_interface.

Does this answer your question?

byeongkyu, 2021-10-06:
@destogl The rate at reading and writing is determined by controller_manager <update_rate> parameter. And controller_manager reads whole URDF <robot_description> at launch. I don't know how can I set the different rates per each hardware_interfaces.

cco-seasony, 2022-03-04:
Hey, I have been reading the comments and @bmagyar mentioned that it is possible to set the namespace to the controller manager when loaded as a node.
Are there any solutions for adding a namespace when using gazebo_ros2_control? I am struggling to load multiple robots with independent controller_managers