ros-controls/ros2_control #573: Dockerfiles in release branches all pull the same repos file

jaron-l, 2021-11-06:
This is kinda low priority because, as I discuss below, the repos files are currently identical but I thought I'd raise it to document the problem.

**Describe the bug**
When building the source Dockerfiles, the repos file that it pulls to clone the separate ros packages is always the master version. This means that the foxy branch version of the Dockerfile is pulling the master (currently galactic) repos file. This currently doesn't matter as they are identical on those two branches but could potentially cause a bug in the future should they diverge.

**To Reproduce**
1. Build the [foxy branch source Dockerfile](https://github.com/ros-controls/ros2_control/blob/98f9978688df935058aae89d8a66db7eb69c2ce6/.docker/source/Dockerfile#L8) and see that it pulls the master branch repo file.

**Expected behavior**
The foxy branch (and all supported branches) should pull the repos file from their branch.

**Environment (please complete the following information):**
Docker

**Additional context**
I think this bug highlights a slightly larger issue: there are ros version specific references in CI and the Dockerfiles that need to change when a ros version changes. I see that in the CI files, you just hard coded the ros version. This will mean that every time you branch from master (or whatever you decide to call the branch tracking rolling), you'll need to go and change the CI and dockerfile rolling references else they will all pull the wrong ros versions. It *might* be easier to have the branch name always match the ros version (including rolling) so that you can just use that as the variable in CI and dockerfiles.

destogl, 2021-11-09:
> **Describe the bug** When building the source Dockerfiles, the repos file that it pulls to clone the separate ros packages is always the master version. This means that the foxy branch version of [...]

Thank you for the observation! For docker itself, we should probably add flags into `docker compose` script to enable setting different versions. 

> **Additional context** I think this bug highlights a slightly larger issue: there are ros version specific references in CI and the Dockerfiles that need to change when a ros version changes. I see [...]

I agree with you on this, still, this will not change the issue of having different files for each distribution. The reason for this is scheduled build, i.e., GH-Actions only check the default (in our case `master`) branch for scheduled CI execution. TBH I am not 100% sure which files are also used when triggering CI on PRs and push events. So unfortunately, there are not many options there, but having duplication of files. On the other side, this makes the things more explicit and simpler to understand.

Still, if you have any idea how to optimize this, let us know.

jaron-l, 2021-11-10:
> Thank you for the observation! For docker itself, we should probably add flags into `docker compose` script to enable setting different versions.

We already do have the flags for the ROS version. The problem is that depending on the ROS version, you have to pull a certain branch's repo file, which will need a different version. I'll make a pull request to show what I'm proposing. Perhaps that will make it clearer.

> I agree with you on this, still, this will not change the issue of having different files for each distribution. The reason for this is scheduled build, i.e., GH-Actions only check the default (in [...]

I think I could write an example CI configuration that would work for the different ROS versions. I'll try to make a draft pull request to show the conditional checking.

bmagyar, 2021-11-11:
Can't we inject the branch name (or something similar) via args into the container?

jaron-l, 2021-12-08:
@bmagyar, that is what I did in my most recent PR. My concern is that there are several ros distro specific references throughout the ci and repos files that need to be manually changed on release branches. I'm trying to think of a more automated/less error prone way of changing all those references.

PR #592 (2021-12-26): ci: add docker build workflow
* adds docker build workflow
* fixes source dockerfile from pulling the wrong repos file

fixes: #573
commit f9cbc61b (2021-12-26): ci: add docker build workflow (#592)

* ci: add docker build workflow

* add docker github workflow
* fixes source Dockerfile from pulling the wrong repos file

fixes: #573

* cli: break out release and source Dockerfiles in build step

* ci: :build: add buildx and buidx args fixes

refs: #573

* doc: :memo: add docker prebuilt information to README.md
closed 2021-12-26 by bmagyar