ros-controls/ros2_control #591: Add *.pyc to .gitignore?

afrixs, 2021-12-08:
**Is your feature request related to a problem? Please describe.**
For me it's more comfortable to build workspaces using `--symlink-install`. Running controller_manager spawner then creates `.pyc` files inside the repo. Using/modifying many other git repos in my workspaces and using scripts to check whether I forgot to add/commit some changes in repos I modified, this one always alarms about changes...

It's maybe a very subjective request but maybe it would help also others using the repo in a similar way... Thanks!

**Describe the solution you'd like**
Add *.pyc to .gitignore

**Describe alternatives you've considered**
Add __pycache__ to .gitignore

destogl, 2021-12-15:
Hi @afrixs,

it sounds like you have this issue on multiple repositories. Why not add a `.gitignore` to your workspace-root, so can have it all the subfolders? I have it actually in my home-folder so it is applied to all projects I am working on...

closed 2021-12-22 by bmagyar

afrixs, 2021-12-22:
Hi @destogl,
just tried it out, simply adding `.gitignore` to parent folder seems not to affect submodules, however a global gitignore file can be configured using
```
git config --global core.excludesfile ~/.gitignore
echo *.pyc >> ~/.gitignore
```
so that solved the problem. Thanks for pointing me to the correct direction and sorry for opening this issue just because of my limitedness in git API knowledge

destogl, 2021-12-22:
@afrixs no problem! Glad that I could help!

tonynajjar, 2022-09-26:
Happy to see I wasn't the only one annoyed by the *pyc files generated every time I test my code. The user could create his own gitignore but I think it would make sense to have a gitignore for the project (like in [Nav2](https://github.com/ros-planning/navigation2/blob/af72761f626d8a6bd172d3a05ad070672b51f36c/.gitignore#L45)). You wouldn't want anyone pushing *.pyc by mistake, it's especially useful for less experienced contributors.